capacitor-community/bluetooth-le #273: Bug with notifications on iOS

maelp, 2022-01-14:
**Describe the bug**
I'm adding a notifications listener, and the notifications are not sent, but there is a "read" key instead

**To Reproduce**
Steps to reproduce the behavior:
1. create a notification with a callback
2. run the app on iOS
3. many error logs instead of notifications

**Expected behavior**
A notification is sent

**Screenshots**
I receive errors like this in the console

```
Resolve callback not registered for key:  read|00000000-0000-4000-A001-676F75616368|00000003-0000-4000-A001-676F75616368
```

When I log the available keys I have this (correctly registered for "notification", I'm not sure why the plugin receives the requests as "read" instead)
```
Keys: ["notification|00000000-0000-4000-A001-676F75616368|00000003-0000-4000-A001-676F75616368"]
```

**Plugin version:**
- @capacitor-community/bluetooth-le: latest

**Desktop (please complete the following information):**
- OS: iOS latest

**Smartphone (please complete the following information):**
- Device: iPhone 7
- OS: iOS latest

pwespi, 2022-01-16:
The logs about `Resolve callback not registered for key: read...` are expected when using notifications, as iOS uses the same `didUpdateValueFor` callback for reads and notifications.

Do you get any other errors? Can you create a minimal reproduction and share the code?

maelp, 2022-01-16:
No, there's no other error, just those logs, is there a way to remove them on iOS, as there are quite many notifications so this makes it harder to look at the logs

pwespi, 2022-01-22:
No, there is currently no way to remove those logs. 

I'm not able to reproduce this problem. Could you create a minimal reproduction and share the code?

closed 2022-02-22 by github-actions[bot]