capacitor-community/bluetooth-le #292: Virtual beacon / Advertising

kolliss, 2022-02-20:
My goal is to create a virtual beacon, and it doesn't matter what type of beacon it simulates.

Requirements:
1) posibility set UUID in the manufacturer's data (or other proper place)
2) posibility set advertising frequency

This would be happen something like that:
```
const advData = {
  company: '76',
  uuid: '81542ace-2b93-4995-a738-7e135634f891',
}

const options = {
  frequency = 10, // Hz
  txPower = -66, // dBm
  timeout = 3600 * 1000 // ms - stop advetising after 1 hour
}

BleClient.startAdvertising(advData, options)
...
BleClient.stopAdvertising()
```

pwespi, 2022-02-20:
Thank you for your feature request. 

This plugin currently only implements the central role of the Bluetooth Low Energy protocol. The peripheral role is not implemented, therefore advertising is not possible.

Plugins that implement the peripheral role:
* https://github.com/randdusing/cordova-plugin-bluetoothle
* https://github.com/don/cordova-plugin-ble-peripheral

I will leave this feature request open to see if other people are interested as well, but I'm currently not planning to implement the peripheral role in this plugin.

pwespi, 2023-01-14:
It's still not planned to implement the peripheral role in this plugin and the Bluetooth Web API wouldn't even support it. Therefore I'm going to close this feature request.

closed 2023-01-14 by pwespi