capacitor-community/bluetooth-le #296: Set Custom Connection Timeout

asangadev, 2022-03-02:
**Describe the bug**
I'm trying to set a custom connection timeout in my App. 

**To Reproduce**
```await BleClient.connect(deviceId, () => undefined, { timeout: 20000 })```

**Expected behavior**
20 seconds timeout and not the default 5 seconds

**Screenshots**
[image]

**Plugin version:**
- @capacitor-community/bluetooth-le: 1.8.0

**Desktop (please complete the following information):**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Smartphone (please complete the following information):**
- Device: iPhone 12 Pro
- OS: iOS 15.3.1
- Browser: N/A
- Version: N/A

**Additional context**
N/A

asangadev, 2022-03-04:
@pwespi thanks for the reply. Sorry yeah, you are right. Looks good with on connect. See attached. So how this `let timeout = self.getTimeout(call)` works in `startNotifications `?

Can we configure that timeout using the JS API (using options, the same way we configure the connect)? Please advise.

[image]

pwespi, 2022-03-08:
No, setting a custom timeout is not exposed for  `startNotifications`, only for calls like `read` and `write`. Do you run into timeout errors? In my experience, `startNotifications` is very fast. 

Note that `startNotifications` resolves after the subscription has been created, and the values can come any time later until `stopNotifications` is called. So the timeout only applies to the opening of the subscription, not to the transmission of values.

asangadev, 2022-03-08:
@pwespi Thanks for the reply! This is very clear to me now. I will close this. Thanks again!

closed 2022-03-08 by asangadev