capacitor-community/bluetooth-le #297: Change connection priority

aanders77, 2022-03-02:
**Is your feature request related to a problem? Please describe.**
We're going to send firmware updates to a BLE device, and want the transfer to take as little time as possible.

**Describe the solution you'd like**
It would be nice to be able to change connection priority https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestConnectionPriority(int)

Also, being able to request a bigger MTU size might help to increase transfer speed.
https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestMtu(int)

**Describe alternatives you've considered**
The device manufacturer sent us a java-based example application where they used the two mentioned methods to increase transfer speeds. But the firmware is less than 1MB, so tbh I haven't got a chance yet to test if there is a noticable difference.

Arxi, 2023-03-26:
I'd also love to be able to request a bigger MTU on Android for the purpose of OTA firmware updates. Seems like some code for requesting is already present, but not public and used only during service discovery, if I understand correctly:

https://github.com/capacitor-community/bluetooth-le/blob/main/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt#L210

Maybe if we could request higher MTU while connecting to a peripheral, and we'd somehow be able to get the actual MTU number back, that would be sufficient.

gregoiregentil, 2023-03-30:
@Arxi, you are hijacking this thread and what you are saying seems inaccurate to me :-(

MTU and connection priority are two different things. MTU concerns the maximum length of data. On Android, the plugin tries 512 bytes which is the maximum possible by BLE specs (read Device.kt). On iOS, maximum MTU is 185 and I don't see in the plugin code anything to set it up.

The connection priority is going to try to change the min/max internals which is critical for the speed.

gregoiregentil, 2023-03-30:
I would like to see also an option to change connection priority. I have made an ugly hacky patch of Device.kt to force the highest connection priority. I think that a proper implementation will be to add priority level in the options and to bring down to Device.kt. I don't think that it's super complicated but you need to touch all the levels of the plugin stack.

Arxi, 2023-03-31:
@gregoiregentil thank you for thorough explanation. I was referring specifically to the second part of the original issue, see the first post:

>Also, being able to request a bigger MTU size might help to increase transfer speed.
https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestMtu(int)

pwespi, 2023-04-02:
Version 2.2.0 contains two new methods:

- `requestConnectionPriority` (Android only) to change the connection priority
- `getMtu` (Android and iOS) to get the MTU size

Note that requesting a larger MTU is already done under the hood. On iOS this is handled by the OS and there is no API to change it. On Android this is done automatically by this plugin.

closed 2023-04-02 by pwespi

gregoiregentil, 2023-04-03:
Thank you for your hard work. Works on my Android.

Arxi, 2023-04-03:
great job, @pwespi, thank you! `getMtu` works on my Android and iOS too

aanders77, 2023-04-03:
Thanks @pwespi 😊😊