capacitor-community/bluetooth-le #301: startNotifications error "Setting notification failed"

karunt, 2022-03-11:
Writing to a BLE device characteristic and expecting to read notifications written to the characteristic.  Using `BleClient.read(deviceID, serviceID, characteristicID)` resolves and returns valid data from the characteristic.  However, using 
`BleClient.startNotifications(deviceID, serviceID, characteristicID, (value)=> {console.log('value is: ', value)})`
returns error "Setting notification failed".  Don't understand how to fix it.  All I know is that the characteristic I'm trying to read changes in value after the BleClient.write command is issued, and so I'd like to be notified each time the characteristic values change, which is why I thought using BleClient.startNotifications might work better than BleClient.read.  Am I misunderstanding the purpose of the startNotifications api entirely? Appreciate some help.

pwespi, 2022-03-11:
Yes, `startNotifications` is generally there to be notified of changes of the value of a characteristic. However, you can only use it if the peripheral device allows it. Probably your device does not allow it for that characteristic.

You can get a list of services and characteristics with their properties with [`getServices`](https://github.com/capacitor-community/bluetooth-le#getservices). The property `notify` or `indicate` has to be `true` for a characteristic for `startNotifications` to work.

karunt, 2022-03-11:
Thanks for the feedback and tip.  Will revert with questions if I get stuck again.

karunt, 2022-03-12:
BTW, what's the proper way to use getServices?  I tried BleClient.getServices(connectedDeviceID), and I get an error saying 
"Property 'getServices' does not exist on type 'BleClientClass'."  But isn't the recommendation to use each of the functions/apis with BleClient?

pwespi, 2022-03-13:
Yes, you should always use BleClient. 
Which version of the plugin are you using? `getServices` was added in 1.4.0

pwespi, 2022-03-17:
Yes, usually you get the updated data back from the notification. There is a `startNotifications` example in the [usage](https://github.com/capacitor-community/bluetooth-le#usage) section of the readme.

However, it all depends on the logic that is implemented in the peripheral device. A peripheral device can decide to send a completely different value over notifications than as an answer to a read call. It can send a notification every time you write, or just once every minute, or never at all, or really just anything that is implemented on the peripheral device. So you need to know how your specific device handles notifications.

pwespi, 2022-03-17:
I'm going to close this issue, as I don't think it is a problem with this plugin, but has more to do with your specific device.

If you have any problems that are related to the plugin, please let me know and I can reopen the issue.

closed 2022-03-17 by pwespi

karunt, 2022-10-11:
So I updated my npm package for capacitor-community/bluetooth-le to 1.8, and now startNotifications doesn't give an error. The getServices api also shows notification property set as true.  However, it doesn't give me updated data on the desired characteristic like the read api does.  Here's my code for the read api:
BleClient.read(deviceID, baseUUID, characteristicID).then((resp)=>{  const binMsg = new Uint8Array(resp.byteLength);  for (let i=0; i< resp.byteLength; i++){    binMsg[i] = resp.getUint8(i);    console.log('binMsg value: ', binMsg[i]);  }}
Output from the read operation looks like this:
binMsg value: 8binMsg value: 1
binMsg value: 90
binMsg value: 2
binMsg value: 6

Each of the numbers above correspond to a message coded with google protobuf.  I was expecting a similar response from the startNotifications api - to get the same google protobuf encoded response as it changed on the characteristic.  Instead, I get nothing, not even an error message.  Here's my startNotifications code (inside an async function):
await BleClient.startNotifications(deviceID, baseUUID, characteristicID, (resp)=> {  const binMsg = new Uint8Array(resp.byteLength);  for (let i=0; i< resp.byteLength; i++){    binMsg[i] = resp.getUint8(i);    console.log('binMsg value: ', binMsg[i]);  }}).catch((error) => console.log('start notifications error: ', error))
Should I be expecting updated data from startNotifications, or just a signal that something on the particular characteristic has changed, following which the read api needs to be executed to obtain the updated data?  I guess it would be helpful to see some sample output pertaining to the startNotifications api.