capacitor-community/bluetooth-le #306: Problem connecting to Ble device without rescan and manual device selection

codeclinic, 2022-03-24:
Hi,

I'm not sure if this is a bug or an issue with my code.

When I try to connect to a device using requestDevice this shows the available devices and I can then pair/connect with that device and all is good.

```
    await BleClient.initialize();
    if (isAndroid) {
       await BleClient.disconnect(devices.deviceId);
    }

    const device = await BleClient.requestDevice({
       services: services ? services : [],
       optionalServices: optionalServices ? optionalServices : [],
       namePrefix: prefixFilter ? prefixFilter : "",
    });

    await BleClient.connect(device.deviceId, (deviceId) => onDisconnect(deviceId));

    await BleClient.getServices(device.deviceId).then(
       (services) => {
           if (services[0]) {
              //....
           } else {
              //....
           }
       }
    )
```

However, if I save the device ID and then try to directly connect with that device using getDevices (rather than scanning and manually connecting) it always fails with the following console output:

> Uncaught (in promise) Error: Device not found. Call "requestDevice", "requestLEScan" or "getDevices" first.

The code I use is this:

```
    await BleClient.initialize();
            
    if (isAndroid) {
       await BleClient.disconnect(devices.deviceId);
    }
            
    await BleClient.getDevices(devices.deviceId);
    await BleClient.connect(devices.deviceId, (deviceId) => onDisconnect(deviceId));
```

***For clarification:*** I want to be able to search for available devices and connect to the device the first time the app is opened. Then, save the device ID and use this ID to connect to the device directly using getDevices from that point onwards. Likewise if the app is closed and re-opened I need to be able to take the stored device data and connect with that device directly without the whole scan and manual selection process.

I really don't understand what I'm missing and would appreciate any suggestions / pointers.

Thanks.

pwespi, 2022-03-28:
Yes, `getDevices` is meant for that. However, on the web platform, it requires [`getDevices`](https://developer.mozilla.org/en-US/docs/Web/API/Bluetooth/getDevices) from the Web Bluetooth API, which is still experimental and requires a flag. So on web it's not really usable for production. See the [platform support](https://github.com/capacitor-community/bluetooth-le#platform-support) table in the readme.

Have you tried on Android and iOS?

codeclinic, 2022-03-28:
Thanks for the response. I already set the relevant flags in chrome as detailed via the platform support references.

I've not been able to test on Android or iOS just yet.

pwespi, 2022-04-10:
In my tests this works on Android and iOS. On web this plugin is limited by what the Web Bluetooth API can do. So currently I don't see anything that can be improved in the plugin, so I'm closing this issue for now. If it doesn't work for you on iOS or Android, please let me know and I can reopen.

closed 2022-04-10 by pwespi