capacitor-community/bluetooth-le #308: RequestDevice, the deviceID appear on Android.

ohlive, 2022-03-25:
**Describe the bug**
Its not really a bug but i want to know how i can fix that because is not really pretty.

For beginning, im a junior developper so my question can be simple for you.

I use our code to request device. 

```
await BleClient.initialize()
    ble_enabled.value = await BleClient.isEnabled()
    await BleClient.startEnabledNotifications((enabled: boolean) => {
      ble_enabled.value = enabled
    })
    const device = await BleClient.requestDevice({
      name: 'orphee'
    })
```
When i found my device ( and its work well) on a screen we have the device ID and the name .

[image]

Its possible to not display this device ID?

Other question?

I do that to connect to this device.
`
await BleClient.connect(device.deviceId, deviceId => onDisconnect(deviceId))`
  
I possible to connect directly in background of my application?

Thanks for your time.

**Plugin version:**
- @capacitor-community/bluetooth-le: 1.1.1

codeclinic, 2022-03-26:
If I understand correctly, you can connect to a device directly via it's deviceId using getDevices.

ohlive, 2022-03-28:
Oh thanks, i look that today. for getDevices.

But i would like know equally how i can search device (requestDevice) without displayed the deviceID is possible?

Thanks

codeclinic, 2022-03-28:
I don't think so.

P.s. I''m currently having issues with getDevices, so I'd be interested to see how you get on...

pwespi, 2022-03-28:
> But i would like know equally how i can search device (requestDevice) without displayed the deviceID is possible?

No, it's not possible to hide the deviceId with `requestDevice`. However, you can use [requestLEScan](https://github.com/capacitor-community/bluetooth-le#requestlescan), which returns the scan results in a callback, and then you can implement your own dialog in JavaScript if you like.

ohlive, 2022-03-31:
Ok thanks for the response. For the moment i will use Request device, maybe in the future, i try this.

pwespi, 2022-04-10:
I'm closing this as I think the questions have been answered.

closed 2022-04-10 by pwespi