capacitor-community/bluetooth-le #316: isLocationEnabled() is returning true even if app has no permissions

Fabchick, 2022-04-12:
**Describe the bug**
On my Android phone (Android 11, api 30, OnePlus 7t pro) a call to (awaited) `BleClient.isLocationEnabled()` is always returning `true`. 

**To Reproduce**
Steps to reproduce the behavior:
1. Delete app storage OR set location permission in app settings to denies
2. Open app
3. Call `BleClient.isLocationEnabled()` as first `BleClient` call
4. Wait for Promise result
5. See `result === true`

**Expected behavior**
When app has no location permission set or has denied it, then `BleClient.isLocationEnabled()` returns false.

**Plugin version:**
tested with:
- @capacitor-community/bluetooth-le: [1.6.0]
and 
- @capacitor-community/bluetooth-le: [1.8.0]
(here I also used the new {androidNeverForLocation: true} and the suggested settings on the android files.)

**Additional context**
I want to show the user an explaining dialog before showing the location permission prompt from android. Therefore I call `BleClient.isLocationEnabled()` to show it or not. 

I ask myself: Should some `BleClient` method be called prior to `isLocationEnabled()` to ensure it is working correctly?

pwespi, 2022-04-24:
`isLocationEnabled` is meant to report whether the location service of Android is turned on or off, not whether the app has permission to use the location, see https://developer.android.com/reference/android/location/LocationManager#isLocationEnabled().

If you turn off location in the Android system settings, does `isLocationEnabled` still return `true`?

Fabchick, 2022-04-27:
You are totally right. I misunderstood `isLocationEnabled`.
Thank you very much for clarification!

I turned off the location of my Android system and `isLocationEnabled` returned `false`. It works fine.

closed 2022-04-27 by Fabchick