capacitor-community/bluetooth-le #343: Write failure and 'characteristic not found' on android

danroot, 2022-06-03:
**Describe the bug**
I have written an ionic framework app that uses capacitor-community/bluetooth-le to read and write from a device.  It works fine on ios. On android it will read all characteristics successfully.  However, some writes will fail, and then after that the same reads that were previously working fail with the error 'Characteristic not found'.  I know the characteristic exists and works because they previously work, and work fine on ios.  Once in this state, the only thing I've found that can resolve is to "Forget device" in android BT settings, then re-pair and try again.

**To Reproduce**
Reproducing the behavior will be difficult as the device is a custom device.  I'm hoping the logs and code below will provide enough context to help identify the issue.  When looking at the logs, note there is an authorization step where the user may be prompted to enter a password to a characteristic on the device before it lets them write to the protected characteristics.

**Expected behavior**
Expected behavior is for Android app to work as well as the ios app.  I need to know why the write is failing on android, when the same code works as expected on ios, AND how to prevent it from getting stuck in the 'Characteristic not found' state in the event of future failed writes.

**Screenshots**
NA

**Plugin version:**
- @capacitor-community/bluetooth-le: 1.8.2

**Smartphone (please complete the following information):**
- Device: Nexus 6
- OS: Android 7.1.1
- Browser: na/chrome 
- Version 7.1.1

**Additional context**
[bleissue.txt](https://github.com/capacitor-community/bluetooth-le/files/8834665/bleissue.txt)

pwespi, 2022-06-06:
Thank you for using this plugin.

Unfortunately, I don't understand what's going on from the logs. 

Have you tried with several Android phones?

If there is an authorization step involved, your can try using the explicit methods [`createBond`](https://github.com/capacitor-community/bluetooth-le#createbond) and [`isBonded`](https://github.com/capacitor-community/bluetooth-le#isbonded) (on Android only).

danroot, 2022-06-10:
After testing with a newer device, I'm pretty sure this was caused by flaky bluetooth on the Nexus 6.  Same code works fine on Pixel 5 with newer android.

closed 2022-06-10 by danroot

pwespi, 2022-06-10:
Good to know. Thanks for reporting back.