capacitor-community/bluetooth-le #353: Error: Writing characteristic failed

Sameerhesta, 2022-06-24:
Hi,

I am trying to write some data on characteristics using BleClient.write() and I am getting the error **Writing characteristic failed** [Ref](https://github.com/capacitor-community/bluetooth-le#write) 
and some time while trying to start notification or writing Descriptor with writeDescriptor also gives the error **Writing descriptor failed**

I also get the properties using **BleClient.getServices()** where write is true.
 
I am new to this and unable to understand why I failed to write the characteristic.

Is there any way to know the detailed reason for these errors?

pwespi, 2022-06-25:
The best place to find more information is probably the log in Android Studio / XCode.

Sameerhesta, 2022-06-25:
The error I have shared in question was in the logs of Android studio.

But I can't understand the reason of the error, why it is being failed to write the characteristic?

pwespi, 2022-07-13:
I cannot reproduce this. Did you find anything in the logs?

Sameerhesta, 2022-07-18:
Yes,
Actually, the issue was from the device side.
The device was giving **unlikely error** .

closed 2022-07-18 by Sameerhesta

israeloriente, 2025-06-21:
I was having this problem because when using BleClient.write(), it needed to wait more than 30 seconds, which caused the error “Error: Writing characteristic failed”. The data was actually being sent correctly, but the connection was breaking with this error...

How did I fix it?

I removed the await, and it worked…

I changed from:
await BleClient.write()
to:
BleClient.write()
(assuming you’ve already started the notifications before this command)