capacitor-community/bluetooth-le #355: "Invalid Data" Error prevents valid write of empty arraybuffer

danroot, 2022-06-27:
**Describe the bug**
I am using this plugin to write to a device whose spec says that writing an empty value to a given characteristic is necessary to remove a given value.  They have demonstrated this using the nRF Connect Desktop application used to debug bluetooth communication.  When I attempt to write an empty ArrayBuffer via this plugin, I get the error "Invalid Data".  After looking at the code, it looks like a previous fix #131 broke the ability to write an empty value to the device.

**To Reproduce**
Steps to reproduce the behavior:
await BleClient.write(address, service, characteristic, new DataView(new ArrayBuffer(0)));
Observe client fails with "Invalid Data" error.

**Expected behavior**
The plugin should allow some way to write an empty value to a given characteristic without throwing an exception.

**Plugin version:**
- @capacitor-community/bluetooth-le: 6.6.2

**Smartphone (please complete the following information):**
- Device:  iPhone 12 Pro Max
- OS: iOS 15.5

**Additional context**
I suspect the problem is with the error checks on line 248 and 288 here: https://github.com/capacitor-community/bluetooth-le/blob/c27748b51765dcd88e06a8d94269d1ea34f499ec/ios/Plugin/Device.swift
and line 258 and 472 here:
https://github.com/capacitor-community/bluetooth-le/blob/c27748b51765dcd88e06a8d94269d1ea34f499ec/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt

I suspect instead of rejecting if value == "", it should instead write an empty byte array and only reject if the write fails.

danroot, 2022-06-27:
I have verified the lines above were the issue by using a forked copy in my app: https://github.com/danroot/bluetooth-le

Let me know if I should do a PR, but I'm less clear on the reason this was done in the first place, and if there's some other type-checking or flag that should be used instead of my simple fix of just removing the check.

closed 2022-07-17 by pwespi

pwespi, 2022-07-17:
Thank you for this issue and the detailed description. You found the right lines that caused this behavior. There were some more code changes necessary to also correctly allow reading empty values. 

The fix has been released in version 1.8.3.

PR #363 (2022-07-17): Empty values
closes #355