capacitor-community/bluetooth-le #358: iOS - Not receiving call back when enabling startNotifications

rudreshHestabit, 2022-07-06:
**Describe the bug**
1) I am not receiving the callback after start notification in iOS code 

**To Reproduce**
Steps to reproduce the behavior:
1) connect device
2) write value to characteristics 
3) startNotifications() with iOS side
4) observe that we are not getting callback response.

**Expected behavior**
A clear and concise description of what you expected to happen.

Reference Code:

        await BleClient.initialize()
        console.log("Here we are")
        const res = await BleClient.requestDevice()
        console.log("--> res",JSON.stringify(res))      
                
        await BleClient.connect(res.deviceId) 
  
        await BleClient.write(res.deviceId, SERVICE, CHARACTERISTIC, numbersToDataView([0x01, 0x00]), 15000)
        await BleClient.startNotifications(
          res.deviceId,
          SERVICE,
          CHARACTERISTIC,
          (value) => {
            console.log("+++++++++++Notification call back++++++++++")
            this.notifyBLE(value)
          },
        )

**Plugin version:**
"@capacitor-community/bluetooth-le": "^1.8.2. and I tried with 1.4.0

**LOGs

dataWrite 2 bytes
Resolvevalue write|197A1398-719D-48CF-87C2-4D0535CB48B1|10E0C266-B3F3-4FB5-A0DF-3B1570BE06E8 Successfully written value.
⚡️  TO JS undefined
⚡️  To Native ->  BluetoothLe removeListener 821728
⚡️  To Native ->  BluetoothLe addListener 821729
⚡️  To Native ->  BluetoothLe startNotifications 821730
notifyCallback
Set notifications true
Reject callback not registered for key:  write|197A1398-719D-48CF-87C2-4D0535CB48B1|10E0C266-B3F3-4FB5-A0DF-3B1570BE06E8
Resolvevalue setNotifications|197A1398-719D-48CF-87C2-4D0535CB48B1|10E0C266-B3F3-4FB5-A0DF-3B1570BE06E8 Successfully set notifications.

pwespi, 2022-07-13:
Unfortunately I cannot reproduce this. Does it happen all the time or just some times? Have you tried different iOS devices?

rudreshHestabit, 2022-07-14:
@pwespi    func write(_ serviceUUID: CBUUID, _ characteristicUUID: CBUUID, _ value: String, _ writeType: CBCharacteristicWriteType, _ callback: @escaping Callback) {
        let key = "write|\(serviceUUID.uuidString)|\(characteristicUUID.uuidString)"
        self.callbackMap[key] = callback
        guard let characteristic = self.getCharacteristic(serviceUUID, characteristicUUID) else {
            self.reject(key, "Characteristic not found.")
            return
        }
        if value == "" {
            self.reject(key, "Invalid data.")
            return
        }
        let data: Data = stringToData(value)
        self.peripheral.writeValue(data, for: characteristic, type: writeType)
//        if writeType == CBCharacteristicWriteType.withResponse {
//            self.setTimeout(key, "Write timeout.")
//        } else {
//            self.resolve(key, "Successfully written value.")
//        }
        self.resolve(key, "Successfully written value.")
    }``

rudreshHestabit, 2022-07-14:
@pwespi  
 1 In the native plugin code I can not able to write characteristics in iOS 12
 2 if writeType == CBCharacteristicWriteType.withResponse when I comment this part of the code I can able to write the values and not getting the response callback from startNotification please help us

Deval99, 2022-12-19:
Everything is working fine in android, but in iOS, device stops receiving notification response after calling BleClient.getConnectedDevices(), 

and everytime it is throwing error, "Resolve callback not registered for key:  ...." But when BleClient.getConnectedDevices() is not called, it is still working.

I have tried to remove BleClient.getConnectedDevices() from whole project, the app is working fine in iOS.

LOGS :
`⚡️  To Native ->  BluetoothLe addListener 80124203
⚡️  To Native ->  BluetoothLe startNotifications 80124204
.
.
.
Resolve setNotifications|<SERVICE_ID>|<B7__UUID> Successfully set notifications.
⚡️  TO JS undefined

Resolve callback not registered for key:  read|<SERVICE_ID>|<B7__UUID>
⚡️  TO JS {"value":"ba 00 00 04 01 00 00 "}
.
.
// RECEIVED A NOTIFICATION HERE
.
.
.
.
// Now main culprit
//-------------------
⚡️  To Native ->  BluetoothLe getConnectedDevices 80124205
⚡️  TO JS {"devices":[{"deviceId":"-------","name":"--------"}]}
//-------------------
.
.
.
.
.
⚡️  To Native ->  BluetoothLe read 80124206
Resolve read|<SERVICE_ID>|<7E__UUID> 00 00 00 00 b7 ae 00 00 32 b7 00 cc 00 2f 00 00 
⚡️  TO JS {"value":"00 00 00 00 b7 ae 00 00 32 b7 00 cc 00 2f 00 00 "}
.
.
.
.
.

⚡️  To Native ->  BluetoothLe write 80124207
Resolve write|<SERVICE_ID>|<C7__UUID> Successfully written value.
⚡️  TO JS undefined

Resolve callback not registered for key:  read|<SERVICE_ID>|<B7__UUID>
.
.
.
.
.

// FAILS TO GIVE NOTIFICATION HERE

.
.
.
3 seconds time gap
.
.
.

⚡️  To Native ->  BluetoothLe stopLEScan 80124208
Stop scanning.

⚡️  TO JS undefined`

pwespi, 2023-04-02:
There is a known issue that notifications stop working after `getConnectedDevices` see https://github.com/capacitor-community/bluetooth-le/issues/460.

Without using `getConnectedDevices`, is the issue described here still happening?

bitforcesrl, 2023-04-07:
i've spent HOURS of debugging trying to figure out which was the problem of notifications not receive, and it was the getConnectedDevices call.
We use this method because some times on live reload the app is reloading but the device is still connected to the phone and if we click connect again we have some issue.
Then we have multiple devices that can be connected to the phone and we need to get the list of them
I think that this is an utility that should be fixed

pwespi, 2023-04-08:
Version 2.2.1 contains a fix for a bug where getConnectedDevices or getDevices stopped existing notifications.

Is the original issue described here still happening in 2.2.1?

closed 2023-05-13 by github-actions[bot]