capacitor-community/bluetooth-le #359: Connection timeout error on first attempt at connecting to BLE peripheral

karunt, 2022-07-08:
**Describe the bug**
Using Android for app development.  When connecting to a requested device, first attempt results in a connection timeout error.  2nd attempt results in Pairing dialogue.  After selecting the "Pair" option from this dialogue, app is able to make a successful connection with peripheral device.  After making successful connection, if I disconnect the device and try to reconnect to it, similar behavior happens (first attempt leads to connection timeout error, 2nd attempt leads to Pairing dialogue, followed by successful connection).  In the first connect attempt, I don't get the Pairing dialogue at all.

This seems to be a recent occurrence.  Wasn't an issue till May 2022.  Not sure what changed since.  Here's the code I'm using to request device and connect to it:

```
  async connectToDevice(){
    try{
      this.connectedDevice = await BleClient.requestDevice();
      await BleClient.connect(this.connectedDevice.deviceId);
      return this.connectedDevice;
    } catch(error){
      console.log('connection error: ', error);
        if(this.connectedDevice){
          const device = this.connectedDevice.name ? this.connectedDevice.name : this.connectedDevice.deviceId;
          throw new Error('Unable to connect to ' + device + '. Select another Bluetooth device.');
        } else{
            throw new Error();
        }
    } finally{
        delete this.connectedDevice;
    }
  }
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I would expect the Pairing dialogue to appear on every attempt to connect a peripheral device (ie, after selecting a device from the list of devices that appears as a result of BleClient.requestDevice()

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Plugin version:**
- @capacitor-community/bluetooth-le: ^1.8.0

**Desktop (please complete the following information):**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Smartphone (please complete the following information):**
- Device: Samsung Galaxy Tab A7
- OS: [e.g. iOS8.1]
- Browser [e.g. stock browser, safari]
- Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

pwespi, 2022-07-13:
Thank you for the detailed issue. Unfortunately I cannot reproduce it, so I don't know how to debug it. Have you tried different Android devices?

closed 2022-08-13 by github-actions[bot]

Utzel-Butzel, 2022-12-20:
I've had an similar issue where the first `read()` is causing the pairing request, but immediatly times out. After I confirm the pairing request I can make `read()` requests without any error. My quick and dirty workaround is calling a random read until the pairing request is confirmed.

```
const waitForPairingRequest = async (device) => {
    try {
      await BleClient.read(
        device.deviceId,
        DEVICE_DATA_SERVICE,
        CHARACTERISTIC
      );
      return true;
    } catch (error) {
        await new Promise((r) => setTimeout(r, 5000));
        await firstTry(device);
      }
    }
};

await waitForPairingRequest(device);
```