capacitor-community/bluetooth-le #365: BleClient.requestDevice case sensitivity issues on Android

ecc521, 2022-07-18:
Calling 
```
await BleClient.requestDevice({
    services: ['___lowercaseid___']
})
```

fails to find any devices on Android, while 
```
await BleClient.requestDevice({
    services: ['___lowercaseid___'.toUpperCase()]
})
```

succeeds. 

This is a significant difference from the web bluetooth API, which requires UUIDs to be lowercase, and throws an error if they are not. Additionally the sample UUID ```0000180d-0000-1000-8000-00805f9b34fb``` is lowercase.

**Expected behavior**
Either UUIDs should be case insensitive, or requestDevice should error if incorrect casing is passed. 

**Plugin version:**
- @capacitor-community/bluetooth-le: 1.8.2

**Desktop:**
- OS: macOS
- Browser Chrome
- Version 12.4

**Smartphone:**
- Device: OnePlus
- OS: Android 10
- Browser Chrome
- Version 7T

**Additional context**
Phone was connecting to an Arduino 33 BLE.

pwespi, 2022-07-18:
Thank you for using this plugin. Unfortunately, I cannot reproduce this with my devices. For me both uppercase and lowercase works on Android. As you mentioned, could it be that some fluke circumstances have occurred?

closed 2022-07-18 by ecc521

ecc521, 2022-07-18:
Based on further testing, it appears the initial behavior was not the result of a problem, and instead fluke circumstances.