capacitor-community/bluetooth-le #368: Error using with Quasar

masiama, 2022-07-19:
**Describe the bug**
Using the BLE plugin with the Quasar framework crashes the app on launch with such an error
```
SyntaxError: The requested module '/src-capacitor/node_modules/throat/index.js?v=0f0edd62' does not provide an export named 'default' (at queue.ts:1:8)
```

**To Reproduce**
Steps to reproduce the behavior:
1. `npm init quasar` (Create new Quasar app)
    * `What would you like to build? › App with Quasar CLI, let's go!`
    * `Pick Quasar version: › Quasar v2 (Vue 3 | latest and greatest)`
    * `Pick script type: › Typescript`
    * `Pick Quasar App CLI variant: › Quasar App CLI with Vite`
    * `Pick a Vue component style: › Composition API`
    * `Pick your CSS preprocessor: › Sass with SCSS syntax`
    * `Check the features needed for your project: › ` (None)
2. Go to newly created project folder
3. `npx quasar mode add capacitor` (Add capacitor platform)
4. `cd src-capacitor`
5. `npm i @capacitor/core@latest @capacitor/cli@latest` (Update capacitor from v2 to v3.6.0)
6. `npm i @capacitor-community/bluetooth-le` (Install plugin)
7. Open `path/to/project/src/pages/IndexPage.vue`
8. Add `import { BleClient } from '../../src-capacitor/node_modules/@capacitor-community/bluetooth-le';` (Import plugin)
9. Inside `setup () { ... }` add before `return`
    ```ts
    
    const init = async () => {
      await BleClient.initialize();
    }

    onMounted(init);
    ```
Then **for web**
10. `npx quasar dev` (Runs app and opens browser)
11. Error in browser's console

**For android**
10. `npx quasar dev -m capacitor -T android` (Adds android platform to the capacitor, builds the app, opens Android Studio)
11. Run the app in Android Studio
12. Inspect device (e.g. `chrome://inspect/#devices`)
13. Error in browser's console

**Expected behavior**
Plugin initializes successfully

**Screenshots**
[image]

**Plugin version:**
- @capacitor-community/bluetooth-le: 1.8.3

**Desktop:**
- OS: macOS
- Browser chrome
- Version 103.0.5060.114

**Smartphone:**
- Device: OnePlus Nord
- OS: Android 11 Oxygen OS 11.1.11.11.AC01BA
- Browser stock browser?

pwespi, 2022-07-19:
This plugin has a web layer, which is also running when using the plugin on Android. It's in this web layer where the package `throat` is used.

I think you will have to install the plugin also in the root project directory (not only in `src-capacitor`) and then import it as usual `import { BleClient } from '@capacitor-community/bluetooth-le';`

This way `npx quasar dev` and `npx quasar build -m capacitor -T android` work for me.

pwespi, 2022-07-20:
Yes, it looks like the quasar/vite dev server does some magic. I think the problem comes from how quasar structures projects with having two separate package.json and node_modules folders. This is usually not the case for capacitor projects. So I don't think this can be fixed in the plugin (or I don't know how), and it would have to be fixed in quasar.

masiama, 2022-07-21:
Thanks for the help. I solved the problem by installing Quasar into the already set up vite project instead of using Quasar CLI

closed 2022-07-21 by masiama

gbkwiatt, 2022-08-06:
Would be nice if there is some proper solution for it. I am facing same issue currently, but creating vite project and adding quasar to it seems like a workaround and not a solution. 
I do understand it's not an issue with plugin, but quasar. But maybe someone can help to tackle that down ?