capacitor-community/bluetooth-le #382: Read without connection not possible?

sanastasiou, 2022-08-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I cannot read anything , not even rssi when not connected to the bluetooth device.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I would like to be able to read non encrypted characteristics without having to connect to the device.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Scanning and then extracting the values I need via advertisement data.
**Additional context**
Add any other context or screenshots about the feature request here.

pwespi, 2022-08-12:
You can get the advertisement data and RSSI using `requestLEScan`. See [scanResult](https://github.com/capacitor-community/bluetooth-le#scanresult).

The `readRSSI` method is only for connected devices, see 
* https://developer.android.com/reference/android/bluetooth/BluetoothGatt#readRemoteRssi()
* https://developer.apple.com/documentation/corebluetooth/cbperipheral/1519111-readrssi

Reading the value of a characteristic is not possible without connecting first. If you need to transmit data without connecting, the peripheral has to add it to the advertisement data, in which case it will be part of the scanResult linked above.

closed 2022-08-12 by pwespi

sanastasiou, 2022-08-12:
Thanks for the responses. I came to the same conclusions as you did. Just one more question, is this some kind of limitation of the plugin or the way it is implemented or is this BLE spec?

pwespi, 2022-08-12:
This is BLE spec.