capacitor-community/bluetooth-le #384: Newbie Read question

donniekerr, 2022-08-12:
My device is a watch that has many sensors sending data over BLE to the phone. My goal is to listen for anything received from the BLEdevice and produce an MQTT message to send the value to the cloud. 
Currently I see how to getServices, get their characteristics, and call read for each one.  But that only happens once when I call read.
Is it possible to achieve my goal? Or is my goal wrong? What is the process for ensuring that I get all data from the watch as it is generated by the sensors?
I'd appreciate any guidance as to how best to architect the BLE side.  I have an handle on the MQTT producing side.
Thanks,
Donnie

pwespi, 2022-08-12:
This depends on the watch and what the services and characteristics offer that it implements. Typically, these characteristics that hold sensor values also support notifications and you can use [startNotifications](https://github.com/capacitor-community/bluetooth-le#startnotifications) to subscribe to changes of the value of a characteristics. This way you get all the values as long as you are connected without having to use `read`.
The other question is whether the watch has internal storage where it saves the values and offers a way to read out historical data as well as the current value. The reading out of historical data is usually implemented by writing to a control characteristic and receiving the data over notifications, but this depends on the watch and you should check its documentation on how and if that's possible.

closed 2022-08-12 by pwespi

donniekerr, 2022-08-12:
Cool. Thanks! Light bulb went off with the StartNotifications.  Appreciate the help!
Unfortunately, the H-Band docs are in Chinese and not helpful to learn if they have a control characteristic. I do know they store data on watch and sync when you pull down in their app. I can't tell from getServices list of characteristics. I'm reaching out to the H-bands guys.  But again I appreciate you taking the time to educate me!