capacitor-community/bluetooth-le #397: Failing to startNotifications and send some writes in a specific phone (works in most phones and some writes also work in the phone with the issue)

ivo-sousaa, 2022-08-31:
Hi,

First thanks for this plugin!

I'm having a peculiar issue, so I after connecting to my device I need to startNotifications for a specific service and also do some writes for some services, this is all working fine in most phones where we tested (all bellow Android 10 as far as I'm aware, and iPhones)... But in a One Plus phone with Android 12 it's failing to startNotifications and failing to send a specific write request, there are some that are working, resolving in `Characteristic not found`, I've been doing some research and just can't pin point it. It's weird that it works with most phones and only some cases fail for this phone (although always the same ones).
Using version 1.5.0 of your plugin.

Do you have any idea of what I can do to resolve this? If you need anymore information I can try to provide it.

- @capacitor-community/bluetooth-le: [e.g. 1.5.0]

- OS: Android
- Version 12

Cheers.

ivo-sousaa, 2022-09-02:
Hi again,

So in my research I found some things in these two links:
https://www.microchip.com/forums/m/tm.aspx?m=1198746&p=1
https://issuetracker.google.com/issues/228984309

So it seems like some models with Android 12 make an automatic call, so if you make your own calls until it's resolved it always returns an error.

pwespi, 2022-09-11:
Hi

Thank you for using this plugin.

Unfortunately these kinds of issues are very hard to debug. I cannot reproduce the behavior you're describing, so I'm not sure I'll be able to help.

Have you compared the logs between the different devices? Any differences that stand out?

kerueter, 2022-09-13:
Hi,

i also encountered this issue and figured out that this issue happens on almost every Xiaomi or Motorola device which are on Android 12 or higher. Tested this on several devices from my colleagues.

So the BLE discovery works, but while calling `startNotifications` this will be thrown:

```
Unable to initialize setup: Characteristic not found.
at http://localhost/main.09c2e5e8ec33ac26.js:1:176017
at Generator.throw (<anonymous>)
at r (http://localhost/main.09c2e5e8ec33ac26.js:1:365888)
at Q (http://localhost/main.09c2e5e8ec33ac26.js:1:366134)
at v.invoke (http://localhost/polyfills.d6d588c4ebc94178.js:1:7705)
at Object.onInvoke (http://localhost/main.09c2e5e8ec33ac26.js:1:507104)
at v.invoke (http://localhost/polyfills.d6d588c4ebc94178.js:1:7645)
at v.run (http://localhost/polyfills.d6d588c4ebc94178.js:1:2783)
at http://localhost/polyfills.d6d588c4ebc94178.js:1:18575
at v.invokeTask (http://localhost/polyfills.d6d588c4ebc94178.js:1:8389)
```

Works on iOS and also on a Google Pixel 4a and some Samsung devices with Android 11 and 12 though..

Unfortunately i don't have a Xiaomi or Motorola device at home atm to go futher reproducing this behavior, but maybe this information helps a bit.

ivo-sousaa, 2022-09-13:
@pwespi I haven't had a chance to use a phone with the problem again so I wasn't able to compare the logs yet. As soon as I can I'll post info here.

Fabchick, 2022-10-13:
On my app I experienced something very similar. Two phones with Android 12 (Xiaomi Mi 10T Lite and an other one which name I don't know) had connection problems. Digging deeper it seemed like the call to `getServices()` right after `connect()` returned no services for these specific phones, which is necessary for `startNotifications()`. 

> So in my research I found some things in these two links: https://www.microchip.com/forums/m/tm.aspx?m=1198746&p=1 https://issuetracker.google.com/issues/228984309
> 
> So it seems like some models with Android 12 make an automatic call, so if you make your own calls until it's resolved it always returns an error.
> 
> Atm resolved it adding a delay of 1.5 secs. Would love to see a better resolution from someone :D

After I read this I tested what happens if I add an 0.5s delay between `connect()` and `getServices()`. With this delay `getServices()` finally returns services and everything works fine now.

717009629, 2022-11-12:
> On my app I experienced something very similar. Two phones with Android 12 (Xiaomi Mi 10T Lite and an other one which name I don't know) had connection problems. Digging deeper it seemed like the call to `getServices()` right after `connect()` returned no services for these specific phones, which is necessary for `startNotifications()`.
> 
> > So in my research I found some things in these two links: https://www.microchip.com/forums/m/tm.aspx?m=1198746&p=1 https://issuetracker.google.com/issues/228984309
> > So it seems like some models with Android 12 make an automatic call, so if you make your own calls until it's resolved it always returns an error.
> > Atm resolved it adding a delay of 1.5 secs. Would love to see a better resolution from someone :D
> 
> After I read this I tested what happens if I add a 0.5s delay between `connect()` and `getServices()`. With this delay `getServices()` finally returns services and everything works fine now.

I have same problem .

przemekRBC, 2023-03-24:
I had the same problem on OnePlus android phone (Samsung phones worked fine). Calling `BleClient.discoverServices()` just after successful connection to a device fixed the issue and I can read and write to characteristics.

pwespi, 2023-04-02:
Reading through all the answers here again, it sounds like on some Android devices (some OnePlus, Xiaomi and Motorola devices) the service discovery during the initial connection sometimes fails or takes longer. If your app is affected, I suggest to use `getServices` and/or `discoverServices` as mentioned above to make sure the characteristic is available before using it. 

Currently I don't see how this could be fixed in the plugin, as it's an issue with specific Android devices. Therefore I'm closing this issue for now. If anyone has an idea how this could be fixed, let me know and I'm happy to take another look.

closed 2023-04-02 by pwespi