capacitor-community/bluetooth-le #404: [Question] good practices of handling BLE Connection.

qpre, 2022-09-20:
This is not a bug, but a discussion.

I wonder what the best practices are for handling the connection to a BLE device,
do you connect when the app is open and only close the connection when the app closes ? 
do you connect/disconnect every time an action (read/write) is performed ?

the first seems way easier to implement (no need for connection locks etc), but what's the effect on the batteries ?

Nickholas, 2022-10-02:
Good question!!

I was wondering the same. I ask every time to the user about it but it would be more confortable if i could reconnect it automatic.

I mean, every time i need to start reading, i show the prompt to connect the device but if he change or close the mine, he lost the connectiong.

peitschie, 2022-10-04:
The question is definitely much better answered for a specific case, because in general "it depends" 🙂 

Connection duration has less effect on battery life than you might think, as it really depends on what the connection intervals are.

In general, connecting consumes a fair amount of power (as scanning is required) compared to a long running idle connection.

Given no other background context, my recommendations would be:
1. automatic connect on launch
2. keep the connection open while the app is open
3. *measure* the battery impact of your app

This is by far the least error prone and most reliable way to get data with mobile phones 🙂