capacitor-community/bluetooth-le #419: "Write timeout." race condition

yelhouti, 2022-10-11:
**Describe the bug**
In few cases the android plugin rejects with timeout even when the request is sent correctly.
This happens In my Opinion here:
https://github.com/capacitor-community/bluetooth-le/blob/b3af2a2113eff38b808edc63cf4c036ae0221189/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt#L477-L484
Where we resolve before clearing the TimeoutMap.
If the javascript code sends a request right away, TimeoutMap[key] is overridden and we loose the reference, this cause the timeout to not be cleared since we lost the reference, and it's triggered later and rejects

I also think that this is a race condition, since the java sometimes cleared before the other request.

**To Reproduce**
Steps to reproduce the behavior:
1. send a first request
2. send a second request on resolve of the first
3. Hope that the seconds request wins the race
4. See error

**Expected behavior**
Not to reject if no error happens

**Plugin version:**
- @capacitor-community/bluetooth-le: 2.0.0

**Smartphone (please complete the following information):**
- Device: Samsung A52
- OS:Android
- Browser WebView
- Version 105

**Additional context**
Add any other context about the problem here.

pwespi, 2022-10-23:
Thank you for this issue.

Unfortunately I cannot reproduce it.

Are you using the `BleClient` like suggested in the [usage documentation](https://github.com/capacitor-community/bluetooth-le#usage) (and not the `BluetoothLe` plugin class)?

Are you using the undocumented `disableQueue` method (not recommended)?

The plugin is implemented with a queue in the JavaScript layer, if you use `BleClient.write()` without disabling the queue, only one call is active in the Android layer at the time.

Andy3189, 2023-08-07:
We are also running into this issue when streaming a few kilobytes of data with a mtu size of 244 bytes and sending data every 30ms. 
The package is transmitted correctly, but the timer is not cleared, which results in a timeout error.

```
2023-08-07 14:01:09.959  9930-9930  Capacitor/Console       io.ionic.starter                     I  (....js) - Line 12 - Msg: [DFU] will send c0 1c 80 b2 90 42 14 d9 80 1a a0 42 00 db 20 46 87 b2 98 18 3a 46 41 46 fb f7 31 fd 00 28 16 d1 e0 1b 84 b2 b8 44 00 20 05 e0 ed 1c ad b2 f6 1e e8 e7 10 1a 80 b2 01 19 a9 42 06 d8 30 44 22 46 41 46 bd e8 f0 41 fb f7 1a bd 4f f0 ff 30 58 e6 2d e9 f0 41 00 f1 28 04 40 7f 1e 46 08 30 90 46 43 18 00 25 08 88 4f 88 06 9a db 1e 90 b1 c0 1c 80 b2 90 42 12 d9 80 1a b0 42 00 db 30 46 85 b2 99 18 2a 46 40 46 fb f7 27 fd 70 1b 86 b2 a8 44 00 20 05 e0 ff 1c bf b2 e4 1e ea e7 10 1a 80 b2 81 19 b9 42 06 d8 21 18 32 46 40 46 fb f7 14 fd a8 19 85 b2 28 46 24 e6 2d e9 f0 41 00 f1 28 04 40 7f 1e 46 08 30 90 46 43 18 00 25 08 88 4f 88 06 9a db 1e 90 b1 c0 1c 80 b2 90 42 12 d9 80 1a b0 42 00 db 30 46 85 b2 98 18 2a 46 41 46 fb f7 f3 fc 70 to IMU_8e447d: 8EC90002-F315-4F60-9FB8-838830DAEA50
2023-08-07 14:01:09.960  9930-9930  Capacitor/Plugin        io.ionic.starter                     V  To native (Capacitor plugin): callbackId: 105122141, pluginId: BluetoothLe, methodName: writeWithoutResponse
2023-08-07 14:01:09.961  9930-9930  Capacitor               io.ionic.starter                     V  callback: 105122141, pluginId: BluetoothLe, methodName: writeWithoutResponse, methodData: {"deviceId":"FE:BD:B7:68:7E:AC","service":"0000fe59-0000-1000-8000-00805f9b34fb","characteristic":"8ec90002-f315-4f60-9fb8-838830daea50","value":"c0 1c 80 b2 90 42 14 d9 80 1a a0 42 00 db 20 46 87 b2 98 18 3a 46 41 46 fb f7 31 fd 00 28 16 d1 e0 1b 84 b2 b8 44 00 20 05 e0 ed 1c ad b2 f6 1e e8 e7 10 1a 80 b2 01 19 a9 42 06 d8 30 44 22 46 41 46 bd e8 f0 41 fb f7 1a bd 4f f0 ff 30 58 e6 2d e9 f0 41 00 f1 28 04 40 7f 1e 46 08 30 90 46 43 18 00 25 08 88 4f 88 06 9a db 1e 90 b1 c0 1c 80 b2 90 42 12 d9 80 1a b0 42 00 db 30 46 85 b2 99 18 2a 46 40 46 fb f7 27 fd 70 1b 86 b2 a8 44 00 20 05 e0 ff 1c bf b2 e4 1e ea e7 10 1a 80 b2 81 19 b9 42 06 d8 21 18 32 46 40 46 fb f7 14 fd a8 19 85 b2 28 46 24 e6 2d e9 f0 41 00 f1 28 04 40 7f 1e 46 08 30 90 46 43 18 00 25 08 88 4f 88 06 9a db 1e 90 b1 c0 1c 80 b2 90 42 12 d9 80 1a b0 42 00 db 30 46 85 b2 98 18 2a 46 41 46 fb f7 f3 fc 70"}
2023-08-07 14:01:09.965  9930-10744 Device                  io.ionic.starter                     D  resolve: write|0000fe59-0000-1000-8000-00805f9b34fb|8ec90002-f315-4f60-9fb8-838830daea50 Characteristic successfully written.
[...]
2023-08-07 14:01:14.955  9930-9930  Capacitor/Console       io.ionic.starter                     I  (....js) - Line 12 - Msg: [DFU] will send 10 e8 49 81 e8 1c 10 d0 e9 04 20 c1 e9 04 20 10 bc 70 47 4f f0 e0 21 02 20 c1 f8 00 01 70 47 e1 49 08 70 70 47 e1 49 08 60 70 47 70 b5 05 46 eb f7 06 f9 de 4c 28 44 e1 6a 88 42 98 bf ff df 01 20 20 74 eb f7 fc f8 da 4a 28 44 00 21 60 61 c2 f8 44 11 d8 49 08 60 a0 6b d8 49 40 f4 80 00 a0 63 d0 01 08 60 70 bd 70 b5 d1 4c 05 46 d4 4a 02 20 20 74 10 68 0e 46 00 f0 0f 00 03 28 08 bf 01 22 13 d0 10 68 00 f0 0f 00 04 28 08 bf 02 22 0c d0 10 68 00 f0 0f 00 05 28 1b d0 10 68 00 f0 0f 00 06 28 1c bf ff df 01 22 13 d0 94 f8 31 00 94 f8 30 10 01 28 15 d0 28 46 0d f0 04 fb bd 49 60 61 00 20 c1 f8 44 01 61 69 e0 6a 08 44 ba 49 08 60 70 bd bb 48 00 68 10 f0 06 0f 0c bf 08 22 04 22 e3 e7 33 46 28 46 0d f0 bb fa e7 e7 b6 49 4f f4 80 00 to IMU_8e447d: 8EC90002-F315-4F60-9FB8-838830DAEA50
2023-08-07 14:01:14.961  9930-9930  Capacitor/Plugin        io.ionic.starter                     V  To native (Capacitor plugin): callbackId: 105122242, pluginId: BluetoothLe, methodName: writeWithoutResponse
2023-08-07 14:01:14.961  9930-9930  Capacitor               io.ionic.starter                     V  callback: 105122242, pluginId: BluetoothLe, methodName: writeWithoutResponse, methodData: {"deviceId":"FE:BD:B7:68:7E:AC","service":"0000fe59-0000-1000-8000-00805f9b34fb","characteristic":"8ec90002-f315-4f60-9fb8-838830daea50","value":"10 e8 49 81 e8 1c 10 d0 e9 04 20 c1 e9 04 20 10 bc 70 47 4f f0 e0 21 02 20 c1 f8 00 01 70 47 e1 49 08 70 70 47 e1 49 08 60 70 47 70 b5 05 46 eb f7 06 f9 de 4c 28 44 e1 6a 88 42 98 bf ff df 01 20 20 74 eb f7 fc f8 da 4a 28 44 00 21 60 61 c2 f8 44 11 d8 49 08 60 a0 6b d8 49 40 f4 80 00 a0 63 d0 01 08 60 70 bd 70 b5 d1 4c 05 46 d4 4a 02 20 20 74 10 68 0e 46 00 f0 0f 00 03 28 08 bf 01 22 13 d0 10 68 00 f0 0f 00 04 28 08 bf 02 22 0c d0 10 68 00 f0 0f 00 05 28 1b d0 10 68 00 f0 0f 00 06 28 1c bf ff df 01 22 13 d0 94 f8 31 00 94 f8 30 10 01 28 15 d0 28 46 0d f0 04 fb bd 49 60 61 00 20 c1 f8 44 01 61 69 e0 6a 08 44 ba 49 08 60 70 bd bb 48 00 68 10 f0 06 0f 0c bf 08 22 04 22 e3 e7 33 46 28 46 0d f0 bb fa e7 e7 b6 49 4f f4 80 00"}

2023-08-07 14:01:14.972  9930-9930  Device                  io.ionic.starter                     D  reject: write|0000fe59-0000-1000-8000-00805f9b34fb|8ec90002-f315-4f60-9fb8-838830daea50 Write timeout. for c0 1c 80 b2 90 42 14 d9 80 1a a0 42 00 db 20 46 87 b2 98 18 3a 46 41 46 fb f7 31 fd 00 28 16 d1 e0 1b 84 b2 b8 44 00 20 05 e0 ed 1c ad b2 f6 1e e8 e7 10 1a 80 b2 01 19 a9 42 06 d8 30 44 22 46 41 46 bd e8 f0 41 fb f7 1a bd 4f f0 ff 30 58 e6 2d e9 f0 41 00 f1 28 04 40 7f 1e 46 08 30 90 46 43 18 00 25 08 88 4f 88 06 9a db 1e 90 b1 c0 1c 80 b2 90 42 12 d9 80 1a b0 42 00 db 30 46 85 b2 99 18 2a 46 40 46 fb f7 27 fd 70 1b 8 [... cut ...] 8 46 24 e6 2d e9 f0 41 00 f1 28 04 40 7f 1e 46 08 30 90 46 43 18 00 25 08 88 4f 88 06 9a db 1e 90 b1 c0 1c 80 b2 90 42 12 d9 80 1a b0 42 00 db 30 46 85 b2 98 18 2a 46 41 46 fb f7 f3 fc 70
                                                                                                        at returnResult (http://localhost/:770:32)
                                                                                                        at win.androidBridge.onmessage (http://localhost/:745:21)
```

pwespi, 2023-08-14:
closed by #560

closed 2023-08-14 by pwespi

olimexsmart, 2025-04-18:
Hi, I get this error, which seems related: 
`Sending plugin error: {"save":false,"callbackId":"76768166","pluginId":"BluetoothLe","methodName":"write","success":false,"error":{"message":"Write timeout."}}`

what can I do?  The target device receives the data correctly. Of all the different commands that I write, I get this error only on the longest (about 80 characters).

peitschie, 2025-04-19:
@olimexsmart are you able to raise a new ticket for discussing this? It's not 100% apparent to me that it's the exact same issue here.

olimexsmart, 2025-04-28:
Sadly I fixed by ignoring the error, the job was due and I needed to wrap it up. I don'tt have access to the target device anymore (it was an ESP32 programmed using the Arduino IDE)