capacitor-community/bluetooth-le #424: BLE packet loss

leydar, 2022-10-18:
### Discussed in https://github.com/capacitor-community/bluetooth-le/discussions/422

<div type='discussions-op-text'>

<sup>Originally posted by **leydar** October 14, 2022</sup>
The BLE protocol is lossless. As in, it will drop the connection in the event of packet loss (where the resent packet has also been dropped). Yet when I move a device out of range or otherwise occlude it from the app, sequential packets are lost. The connection isn't dropped but there are gaps in the data transfer. 

It looks like these packets are received at the protocol level but don't make it through to the notifications callback. I've tried switching message type between indication and notification but found the same behaviour in both.

The packets are sent every 80ms.

My code for testing is straightforward and I'm using v2.0.1.
```
    await BleClient.initialize({ androidNeverForLocation: true });
    await BleClient.connect(macAddress, onDisconnect);
    await BleClient.startNotifications(macAddress, service, characteristic, cb);
```
</div>

jrobeson, 2022-10-21:
Does it work correctly with apps like nordic's nrfconnect? do you receive ever notification? 

Is it purely an issue with this plugin? If so, did changing the timing affect it? try at say 500ms and work your way down.

leydar, 2022-10-21:
> Does it work correctly with apps like nordic's nrfconnect? do you receive every notification?
> 
> Is it purely an issue with this plugin? If so, did changing the timing affect it? try at say 500ms and work your way down.
I recorded the UI in LightBlue and tried to watch it frame by frame but figured the UI might be throttled. nRF Connect has logging so I was able to see missed packets there as well. Thank you for the suggestion.

It looks like the issue is at the sender rather than with the library. Sorry for the misdirection.

closed 2022-10-21 by leydar