capacitor-community/bluetooth-le #443: Vite Import Error

asudevprj, 2022-11-23:
**Describe the bug**
Import error, see error description below.
The error only happens when using vite. Webpack works just fine

**To Reproduce**
Steps to reproduce the behavior:
1. Create a project with quasar and vite support
2. import { BleClient } from '@capacitor-community/bluetooth-le' 
3. Start dev server
4. Error:
[vite] error while updating dependencies:
Vite Error, /node_modules/.q-cache/vite/spa/deps/@capacitor-community_bluetooth-le.js?v=6173ba5e 
X [ERROR] Could not resolve "@capacitor/core"

    node_modules/@capacitor-community/bluetooth-le/dist/esm/plugin.js:1:31:
      1 │ import { registerPlugin } from '@capacitor/core';
        ╵                                ~~~~~~~~~~~~~~~~~
  You can mark the path "@capacitor/core" as external to exclude it from the bundle, which will
  remove this error.

**Plugin version:**
- bluetooth-le: 2.01
- capacitor 4.0

**Additional context**
Other plugins (e.g. sqlite) work as expected.
bluetooth-le with webpack works as expected

pwespi, 2022-11-27:
Can you provide a repo with a minimal reproduction? I cannot reproduce your exact error. However, another error with the vite dev server in quasar is known (it's related to quasar, and not vite), see https://github.com/capacitor-community/bluetooth-le/issues/368.

asudevprj, 2022-11-28:
I was able to get rid of the error by using the suggestions mentioned in [#368](https://github.com/capacitor-community/bluetooth-le/issues/368),
Thank you for your great work!