capacitor-community/bluetooth-le #447: Resolve callback not registered for key

foofel, 2022-11-29:
Hey, thanks for the awesome plugin, it works like a charm but i found one annoying thing, at least on my device.

**Describe the bug**
Registering for a notification on ios spams the xcode log with
```
Resolve callback not registered for key:  read|03257878-BEB5-4087-AB68-AE6A82E4C4D9|095AAC41-C04D-436F-93C1-8EF1CEA45016
Resolve callback not registered for key:  read|03257878-BEB5-4087-AB68-AE6A82E4C4D9|095AAC41-C04D-436F-93C1-8EF1CEA45016
Resolve callback not registered for key:  read|03257878-BEB5-4087-AB68-AE6A82E4C4D9|095AAC41-C04D-436F-93C1-8EF1CEA45016
Resolve callback not registered for key:  read|03257878-BEB5-4087-AB68-AE6A82E4C4D9|095AAC41-C04D-436F-93C1-8EF1CEA45016
Resolve callback not registered for key:  read|03257878-BEB5-4087-AB68-AE6A82E4C4D9|095AAC41-C04D-436F-93C1-8EF1CEA45016
Resolve callback not registered for key:  read|03257878-BEB5-4087-AB68-AE6A82E4C4D9|095AAC41-C04D-436F-93C1-8EF1CEA45016
```

The code actually works but its impossible to read the log while data is received.

**To Reproduce**
Subscribe to a notification on ios, e.g.:
```
export async function testble() {
    await BleClient.initialize();
    await BleClient.connect("1D2E8511-890B-1D32-C6EA-C838ADFA652A", (deviceId:string) => onDisconnect(deviceId));
    await BleClient.startNotifications(
        "1D2E8511-890B-1D32-C6EA-C838ADFA652A",
        "03257878-beb5-4087-ab68-ae6a82e4c4d9",
        "095aac41-c04d-436f-93c1-8ef1cea45016",
        (value:DataView) => {
            // something with the data
        }
    );
    setTimeout(async () => {
        await BleClient.stopNotifications("1D2E8511-890B-1D32-C6EA-C838ADFA652A", "03257878-beb5-4087-ab68-ae6a82e4c4d9", "095aac41-c04d-436f-93c1-8ef1cea45016");
        await BleClient.disconnect("1D2E8511-890B-1D32-C6EA-C838ADFA652A");
        console.log('disconnected from device', "1D2E8511-890B-1D32-C6EA-C838ADFA652A");
    }, 10000);
}
```
Then watch the xcode debug log.

**Expected behavior**
Should not spam the log output with a wrong message.

**Plugin version:**
- @capacitor-community/bluetooth-le: 1.8.3

**Smartphone (please complete the following information):**
- Device: ipad air 4
- OS: 15.7

**Additional context**
The problem seems to happen in

https://github.com/capacitor-community/bluetooth-le/blob/7eb0a34f2533980f372cb2168cf2a5f6cea2ea86/ios/Plugin/Device.swift#L188

pwespi, 2022-12-26:
Thank you for reporting this. You are right, on iOS there are always `Resolve callback not registered for key` logs when using notifications. I will look into this.

pwespi, 2023-01-02:
This has been fixed in v2.1.0. Also the logs of the plugin now respect the `loggingBehavior` of the Capacitor config of your app (https://capacitorjs.com/docs/config#schema).

closed 2023-01-02 by pwespi