capacitor-community/bluetooth-le #454: Notification freezing my Ionic App

aardrasystems, 2022-12-07:
Hi

I am not sure why this happens or if its a bug, but when I use  "startNotifications" until I preform "stoptNotifications" nothing gets updated on the screen of my Ionic App.

I am working on BLE communication with the Wifi module. The module performs a scan of all SSID and returns the list using notification. Once my app receives the notification, I add the list to a buffer which updates a list on my page, but this page does not update till 5 or 10 second depending on when stoptNotifications is called by a setTimeout.

The WiFi module , can take up to 10 seconds to produce a list, but does it much quicker, if i was to get the data much quicker and if i was to update the display buffer, nothing gets updated till stoptNotifications is called. 

Is there a reason why this happens?

mutasimissa, 2022-12-08:
I don't think it has anything to do with the plugin itself, review your implementation and make sure of the bug type, I will suggest you simulate the Buffer update function you are talking about and see if it runs or also crashes the app, it's very common that you are handling the buffer or its DataView in a wrong way

pwespi, 2022-12-11:
I agree with @mutasimissa.

@aardrasystems if you still think it is a problem with the plugin, can you create a minimal reproduction so we can debug it?

aardrasystems, 2022-12-11:
I think its an angular issue, so i am trying few things

pwespi, 2022-12-26:
Have you found the issue? Can we close this issue?

aardrasystems, 2022-12-26:
Yes this can be closed now

closed 2022-12-26 by pwespi