capacitor-community/bluetooth-le #456: how can I get the proper value when using BLE GATT and Hex Values

ahunter135, 2022-12-08:
Hello everyone, this is my first time here. I have a client who needs me to connect to a BLE device and get readings from it using Javascript. 

I am successfully connecting to the device. But now it is time to get readings. Here is the information they sent me:

"To provide efficient transmission of data at higher speeds a CondensedDecimalArray type can be used. This will be transmitted as an array of 32-bit signed integer values in little endian format each being a hex string of 4 bytes. The first two bytes of the array are a 16-bit unsigned integer value in little endian format. This is the number of readings in the array. Each reading should be read from the array and processed by the client application."
"To interpret each value use the integer value and insert a decimal point num_decimal_places from the end of the value. The num_decimal_places is stored in a characteristic in the calibration service."

So here is my code block:
const result = await BleClient.read(device.device.deviceId, this.CALIBRATION_SERVICE, this.NUM_DECIMAL_PLACES);
this.decimal_places = result.getInt8(0);

BleClient.startNotifications(device.device.deviceId, this.CURRENT_READING_SERVICE, this.CONDENSED_VALUE_CHR, (value) => {});

I am just struggling on how to accurately get a value from this. I see it is a 32 - bit signed integer value in little endian format. So looking at the docs of the plugin, i can use getInt32(0, true)

So my thoughts are this: use getInt16(0, true) and getInt16(1, true)? That would be the first two bytes of the value array to determine how many readings are in the array? Am i on the right track?

Any help is greatly appreciated.

mutasimissa, 2022-12-08:
Do you know what is a DataView and Buffer in Javascript? if not please read about them and make sure you understand them well, I suggest you to start from [here](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/DataView).

ahunter135, 2022-12-08:
> Do you know what is a DataView and Buffer in Javascript? if not please read about them and make sure you understand them well, I suggest you to start from [here](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/DataView).

I'm new to DataView and interacting with Bits in JavaScript. That's why I thought I could reach out for some guidance.

mutasimissa, 2022-12-08:
Ok you should read more and test DataView yourself it will give you another superpower in JS :) 
read this it might be helpful too
https://javascript.info/arraybuffer-binary-arrays

anyways you should try something like this:
`
const result = await BleClient.read(device.device.deviceId, this.CALIBRATION_SERVICE, this.NUM_DECIMAL_PLACES);
for (let i = 1; i <= result.byteLength/8; i++) {
    console.log(result.getInt32(i, true));
}
`

ahunter135, 2022-12-09:
> Ok you should read more and test DataView yourself it will give you another superpower in JS :) read this it might be helpful too https://javascript.info/arraybuffer-binary-arrays
>
> anyways you [...]

Okay cool! So I can just loop through the object like that. If the response is a array of 32 bit signed integers. It would be result.byteLength/32? Or are all responses always byteLength/8?

mutasimissa, 2022-12-09:
ok, please work more with JS DataView, it has some learning curve but it's important to understand how buffers are represented in a DataView

- First of all we are looping on a DataView which has different characteristics of normal arrays or objects, but yes you can use for loop to loop throw its bytes
- I used byteLength which is similar to Array.length but for DataView
- After reviewing the code I wrote I think I'm mistaken :confounded:  so here is a complete step-by-step code that will make more sense
`
// lets' create a buffer that has "room" of 8 bytes
const buffer = new ArrayBuffer(8);

// let's create a DataView of that buffer
const data = new DataView(buffer)

// let's add to the dataview int32 which mean it will occupy the first 4 bytes
// btw, obviously; in16 is made of 2 bytes and int8 is made of a single byte
data.setInt32(0, 123456789, true);
// now offsets 0,1,2,3 are representing 123456789 in  4 bytes 
// we have "room" for another int32 because our buffer is 8 byteLength
// we have to start the offset from 4
data.setInt32(4, 987654321, true);
// what will happen if we try to set another int32 on position 8 ?
// it will throw an error saying that Offset is outside the bounds of the DataView

// now we have two int32 in our DataView
// let's loop, we will start from 0
// in your code maybe you have to ask whoever you integrate the ble service from about the offset of the start
// because sometimes the first bytes are reserved for some ACK or similar signals
//in our case we will start from offset 0 and will set our iterator to i += 4 because we know that each 4 bytes are occupied by int32 
for (let i = 0; i < data.byteLength; i += 4) {
    console.log(data.getInt32(i, true))
}
`

mutasimissa, 2022-12-09:
also please for future discussions open a discussion thread instead of an issue, thanks in advance :)

ahunter135, 2022-12-09:
> also please for future discussions open a discussion thread instead of an issue, thanks in advance :)

Sorry, I am more of a lurker. Never really post much. Thank you. 

Your example is so much more helpful! I totally understand now and can apply this to my issue. I learn better by example than reading so after reading the DataView Docs and your example, it really makes a lot of sense!

Thank you again! Hopefully this post can help others get an understanding too.

closed 2022-12-09 by ahunter135