capacitor-community/bluetooth-le #460: After BleClient.getConnectedDevices() cant receive notifications

Deval99, 2022-12-19:
**Describe the bug**
Everything is working fine in android, but in iOS, device stops receiving notification response after calling BleClient.getConnectedDevices(),

and everytime it is throwing error, "Resolve callback not registered for key: ...." But when BleClient.getConnectedDevices() is not called, it is still working.

I have tried to remove BleClient.getConnectedDevices() from whole project, the app is working fine in iOS.

**To Reproduce**
1. Start notifications
2. call BleClient.getConnectedDevices()
3. Receive a notification, (by writing something)

**Expected behavior**
To receive a notification in a callback defined in BleClient.StartNotification()

**Plugin version:**
- @capacitor-community/bluetooth-le: ^2.0.1

**Desktop (please complete the following information):**
- OS: Mac
- Browser :NA
- Version :

**Smartphone (please complete the following information):**
- Device: iphone6 - OS: 
- Browser : NA
- Version :

**LOGS**

LOGS :
`zap To Native -> BluetoothLe addListener 80124203
zap To Native -> BluetoothLe startNotifications 80124204
.
.
.
Resolve setNotifications|<SERVICE_ID>|<B7__UUID> Successfully set notifications.
zap TO JS undefined

Resolve callback not registered for key: read|<SERVICE_ID>|<B7__UUID>
zap TO JS {"value":"ba 00 00 04 01 00 00 "}
.
.
// RECEIVED A NOTIFICATION HERE
.
.
.
.
// Now main culprit
//-------------------
zap To Native -> BluetoothLe getConnectedDevices 80124205
zap TO JS {"devices":[{"deviceId":"-------","name":"--------"}]}
//-------------------
.
.
.
.
.
zap To Native -> BluetoothLe read 80124206
Resolve read|<SERVICE_ID>|<7E__UUID> 00 00 00 00 b7 ae 00 00 32 b7 00 cc 00 2f 00 00
zap TO JS {"value":"00 00 00 00 b7 ae 00 00 32 b7 00 cc 00 2f 00 00 "}
.
.
.
.
.

zap To Native -> BluetoothLe write 80124207
Resolve write|<SERVICE_ID>|<C7__UUID> Successfully written value.
zap TO JS undefined

Resolve callback not registered for key: read|<SERVICE_ID>|<B7__UUID>
.
.
.
.
.

// FAILS TO GIVE NOTIFICATION HERE

.
.
.
3 seconds time gap
.
.
.

zap To Native -> BluetoothLe stopLEScan 80124208
Stop scanning.

zap TO JS undefined`

pwespi, 2022-12-26:
Thank you for using this plugin.

I can confirm the behavior you're describing, however I don't understand why this happens. I will look into it some more.

What are you using `getConnectedDevices` for? If you want to keep track of connections that you establish in your app code, I suggest to use the `onDisconnect` callback of the [`connect`](https://github.com/capacitor-community/bluetooth-le#connect) method, instead of periodically calling `getConnectedDevices`.

The log about "Resolve callback not registered for key:..." is a known issue, see https://github.com/capacitor-community/bluetooth-le/issues/447, but this is unrelated to getConnectedDevices.

Deval99, 2022-12-26:
@pwespi Thank you for answer, i am using `getConnectedDevices` to know whether the device (only one) is connected or not.

> If you want to keep track of connections that you establish in your app code, I suggest to use the onDisconnect callback of the [connect](https://github.com/capacitor-community/bluetooth-le#connect) method, instead of periodically calling getConnectedDevices

i did the same thing

pwespi, 2023-01-29:
> Hi. I have a BLE button, that stops beaconing after it is bonded to iOS to safe the power consumption. So i have no other way to get deviceId to use connect() after app starts, then periodicaly use getConnectedDevices(). Do you have any idea what else could i use?

Why do you need to periodically call `getConnectedDevices`? You can call it once, get the deviceId, and then use that.

bitforcesrl, 2023-04-07:
i've spent HOURS of debugging trying to figure out which was the problem of notifications not receive, and it was the getConnectedDevices call.
We use this method because some times on live reload the app is reloading but the device is still connected to the phone and if we click connect again we have some issue.
Then we have multiple devices that can be connected to the phone and we need to get the list of them
I think that this is an utility that should be fixed

pwespi, 2023-04-08:
This has been fixed in 2.2.1.

closed 2023-04-08 by pwespi