capacitor-community/bluetooth-le #463: No scan result when filter by service UUID

wenge8n, 2022-12-20:
**Describe the bug**
When do requestLEScan with empty services array, it finds the nearby devices correctly. But when I scan with service filter(I tried to add 0x1800, 0x180f etc), there is no scan result. Do you have any idea why?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
It should show the devices with the 0x1800 service.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Plugin version:**
- @capacitor-community/bluetooth-le: 2.0.1

**Desktop (please complete the following information):**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Smartphone (please complete the following information):**
- Device: Huawei
- OS: Android 8.0
- Browser [e.g. stock browser, safari]
- Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Deval99, 2022-12-21:
@wenge8n 

Are you using correct format ? 
https://github.com/capacitor-community/bluetooth-le#uuid-format

Deval99, 2022-12-21:
@wenge8n 

import { numberToUUID } from '@capacitor-community/bluetooth-le';

const myService = numberToUUID(0x180d);

BleClient.requestLEScan(
          { services: [myService] }, ()=>{});

wenge8n, 2022-12-21:
@Deval99 yes, I did like that. I passed numberToUUID(0x180d)

Maybe the filter not working with the 16 bit UUID such 0x1800 etc because all ble devices have this same UUID?

peitschie, 2022-12-22:
@wenge8n out of interest, have you tried looking at the advertising packet with an app such as [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) to see if the service you are filtering on is actually in the packet?

If you do a scan in nrf Connect and tap the item itself (not the CONNECT button), it will show you the details included in the advertising payload.

BLE is a little bit confusing in this regard in that a device can support a service without advertising them in the advertising payload.

wenge8n, 2022-12-22:
@peitschie You are right. I tried to scan with the service UUID in the 'Complete list of 128-bit Service UUIDs' and the filter works! The scan shows the device correctly!
Thank you for your help.

closed 2022-12-22 by wenge8n

beckmx, 2023-04-26:
Hello guys sorry for commenting on a closed issue, but I have a similar problem, if I do this:

```
const BLE_KEYBOARD_SERVICE = "6E400001-B5A3-F393-E0A9-E50E24DCCA9E";
try {
          await BleClient.initialize();

          await BleClient.requestLEScan(
            {
              services: [BLE_KEYBOARD_SERVICE],
            },
            (result) => {
              console.log("received new scan result", result);
            }
          );

          setTimeout(async () => {
            await BleClient.stopLEScan();
            console.log("stopped scanning");
          }, 15000);
        } catch (error) {
          console.error(error);
        }
```

I am not able to find anything, is my query mal-formed?

peitschie, 2023-04-27:
@beckmx is the service in the advertising packet or scan response? Also, are you scanning with the app in the foreground (I presume so, but thought it worth asking)

Using nrf Connect to view the advertising packets and scan responses is a great place to start with diagnosing this.

beckmx, 2023-04-27:
@peitschie thanx for the hint, actually I realized that the device didn’t advertise the service uuid, I had to do it via normal discovery, but in either case it also didn’t work with connection by name, only old normal discovery, I will check that app you said