capacitor-community/bluetooth-le #466: getServices() won't update if new (dynamic) Characteristics are notified

francescodoffizi, 2022-12-30:
**Describe the bug**
I've got a device that expoexposes characteristics after a write action (a pairing functionality), but on Bluetooth-Le when I'm connected to a device I won't get any new services / characteristics discovery notification, nor we do have some "refresh" functionality as the refreshDeviceCache available in don/cordova-plugin-ble-central.

**To Reproduce**
You need a device that exposes dynamic services and characteristics
1. Connect to de BLE device
2. Call getServices()
3. Trigger the BLE action that makes new services / characteristics available
4. Call again getServices()
5. No new service / characteristics are shown by getServices() response

**Expected behavior**
New services and characteristics returned from getServices() method

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Plugin version:**
- @capacitor-community/bluetooth-le: 2.0.1

**Desktop (please complete the following information):**
- Not relevant

**Smartphone (please complete the following information):**
- Device: Ulefone 9P
- OS: Android 10
- Browser Not relevant
- Version Not relevant

francescodoffizi, 2022-12-30:
It seems that the plugin doesn't implement Device Changed characteristic management. This is also mentioned in the Apple Accessory Design Guidelines document.

23.12.2 Generic Attribute Profile Service
The accessory must implement the Service Changed characteristic only if the accessory has the ability to change its services during its lifetime.

The device may use the Service Changed characteristic to determine if it can rely on previously read (cached) information from the device. See the Bluetooth 4.0 specification, Volume 3, Part G, Section 7.1.

pwespi, 2023-01-02:
In version 2.1.0 there is now a new `discoverServices` method that can be used to refresh the services of your peripheral if they change. After calling `discoverServices` the `getServices` call should return the updated list.

closed 2023-01-02 by pwespi