capacitor-community/bluetooth-le #470: Bluefy - notifications support

Dima1322, 2023-01-02:
**Describe the bug**
Notifications are not working in Bluefy browser on ipad 10.2 (2021) - iPadOS 16.1 
Bluefy browser: https://apps.apple.com/us/app/bluefy-web-ble-browser/id1492822055
Read/Write operations work.
In the native version (generated using capacitor) of the same app notifications work well.
Web Bluetooth API notifications also work (for testing was used this sample: https://googlechrome.github.io/samples/web-bluetooth/notifications.html). 

**Plugin version:**
"@capacitor-community/bluetooth-le": "^2.0.1"

Dima1322, 2023-01-06:
I have also tested this on iPhone 13 pro with iOS 16.2 . The same issue occurs.

pwespi, 2023-01-07:
Thank you for the issue. I was able to reproduce it.

In the Bluefy Bluetooth logs, I see the following error:
`StartCharacteristicNotifications: Request was rejected.`

However, I have not idea why this happens and don't know how to debug Bluefy. I contacted the Bluefy support. Hopefully they can help.

y3nd, 2023-03-24:
Same issue here.

Any news from Bluefy ? The war might have impacted their activity..

AurelienGasser, 2023-06-01:
@pwespi any update?

dugrema, 2024-02-05:
This is still not working with iPhone 7 (iOS 15.8.1), Bluefy 3.9.0.

I was able to avoid the error (`StartCharacteristicNotifications: Request was rejected.`) by adding both the notification and indicate flags to the characteristic description in the peripheral. But the notifications are still not being processed.

If you do `Object.getOwnPropertyNames(Object.getPrototypeOf(characteristic))` on the characteristic object, the one in bluefy (unlike Chrome) does not have the `oncharacteristicvaluechanged` event handling property. I'll venture that the event driven notification feature is simply not implemented or works very differently... 

I'm using a RaspberryPi Pico W as peripheral. The same javascript code works with Chrome 116+ on Mac OS 10.13 and Windows 10. 

The readValue and writeValueWithResponse operations are working fine for me in Bluefy.

rebtop, 2024-02-27:
Is Bluefy a viable app considering that they don't respond?

curlyz, 2024-04-15:
Hey guys, just come across this. I can help if there is sample code. I have use BlueFy for 2 Bluetooth products using ESP32. Work normal for me. Here is things to watch out
- Make sure you have location service enabled for the device
- For the bluetooth setup process, have a delay between each call, requestDevice, findService, findCharacteristic, startNotify ... (I use 500ms delay, this is due to WebBluetooth problem)

rebtop, 2024-06-04:
Can you explain how you integrated BlueFy into your WEB site?I have been trying to find out for a while now. BobSent from my iPhone