capacitor-community/bluetooth-le #472: Not disconnecting in iOS

Deval99, 2023-01-04:
**Description**
When calling BleClient.disconnect(), it resolves but doesn't actually disconnect. exiting app doesn't help, user needs to manually turn bluetooth off and wait for 7-10 seconds to actually disconnect. It is happening only in iOS, not in Android.

**To Reproduce**
1. BleClient.connect(<DEVICE_1>)
2. BleClient.disconnect(<DEVICE_1>)

**Expected behavior**
Actually disconnect the device, but it falsely resolves the promise and don't actually does disconnect.

**Plugin version:**
- @capacitor-community/bluetooth-le: 2.1.0 (Same was happening in 2.0.1)

**Desktop**
OS: macOS Monterey
Browser :NA
Version : version 12.4

**Smartphone**
Device: iphone12 mini **and** iPhone 11 Pro Max
OS: iOS
Browser : NA
Version : 15.6.1 (iphone12 mini) **and** 15.7 (iPhone 11 Pro Max)  

**Additional context**
[Another issue i reported](https://github.com/capacitor-community/bluetooth-le/issues/460), Which mentioned calling BleClient.getConnectedDevices() stops notifications (it was related with callback, it was receiving notifications but failed to call callback). 

Are 2 issues related internally ?

pwespi, 2023-01-04:
I don't think the two issues are related.

Could it be that the connection to the peripheral was established in another app or by the OS? In that case, a disconnect call will only disconnect the peripheral for your app, but the actual BLE connection will remain. The is the default behavior in iOS. If your app didn't establish the connection, it does not have the rights to end it.

Deval99, 2023-01-05:
Thanks for response,

I am connecting with same app, and disconnecting. But i have to wait at least 20-22 seconds before disconnecting, or else it stays connected for remaining time. (in short, the app wants to stay connected for at least 20-22 seconds, if you do anything in the app, for example for a minute, it disconnects perfectly)

I have checked now, same issue is happening for "cordova-plugin-ble-central": "1.2.4", so it have no problem with this specific plugin. can we do anything about that ?

pwespi, 2023-01-08:
If it also happens with other apps / plugins, I don't think there is anything we can do about it.

marcelowolfsmartindustries, 2023-01-26:
I think that this issue is related with this one:
https://github.com/capacitor-community/bluetooth-le/issues/482