capacitor-community/bluetooth-le #482: iOS - Failed to encrypt the connection, the connection has timed out unexpectedly.

marcelowolfsmartindustries, 2023-01-20:
**Describe the bug**
First time connects to BLE successfully. Next times gives the error:
error:  Failed to encrypt the connection, the connection has timed out unexpectedly.

**Expected behavior**
Sucessfully connection.

**Screenshots**
[image]

**Plugin version:**
- @capacitor-community/bluetooth-le:2.0.1

**Smartphone (please complete the following information):**
- Device: Iphone 11 PRO
- OS: iOS 16.1.2
- Browser
- Version

pwespi, 2023-01-23:
Hi

Thank you for using this plugin.

Can you share the logs you get from Xcode? Are you able to connect to the device with another app, e.g. nRF Connect?

marcelowolfsmartindustries, 2023-01-27:
I notice that, this happens after bounding.

Xcode logs when success:

```
⚡️  BluetoothLe - Resolve initialize BLE powered on
⚡️  TO JS undefined
⚡️  To Native ->  BluetoothLe isEnabled 104953965
⚡️  TO JS {"value":true}
⚡️  To Native ->  BluetoothLe getConnectedDevices 104953966
⚡️  TO JS {"devices":[]}
⚡️  To Native ->  BluetoothLe addListener 104953967
⚡️  To Native ->  BluetoothLe requestLEScan 104953968
⚡️  BluetoothLe - Resolve startScanning Scan started.
⚡️  TO JS undefined
⚡️  BluetoothLe - New device found:  GLAMM
⚡️  TO JS {"txPower":127,"localName":"GLAMM","uuids":[],"rssi":-73,"device":{"name":"GLAMM","deviceId":"FEE99B3A-8E41-43CB-7EAF-8B1B94A46EBC"}}
⚡️  To Native ->  BluetoothLe removeListener 104953969
⚡️  To Native ->  BluetoothLe stopLEScan 104953970
⚡️  BluetoothLe - Stop scanning.
⚡️  TO JS undefined
⚡️  To Native ->  BluetoothLe removeListener 104953971
⚡️  To Native ->  BluetoothLe addListener 104953972
⚡️  To Native ->  BluetoothLe connect 104953973
⚡️  BluetoothLe - Connecting to peripheral <CBPeripheral: 0x283a30680, identifier = FEE99B3A-8E41-43CB-7EAF-8B1B94A46EBC, name = GLAMM, mtu = 0, state = disconnected>
⚡️  BluetoothLe - Connected to device <CBPeripheral: 0x283a30680, identifier = FEE99B3A-8E41-43CB-7EAF-8B1B94A46EBC, name = GLAMM, mtu = 23, state = connected>
⚡️  BluetoothLe - Resolve connect|FEE99B3A-8E41-43CB-7EAF-8B1B94A46EBC Successfully connected.
⚡️  BluetoothLe - Connected to peripheral. Waiting for service discovery.
⚡️  BluetoothLe - didDiscoverServices
⚡️  BluetoothLe - didDiscoverCharacteristicsFor 1 1
⚡️  BluetoothLe - Resolve connect Connection successful.
⚡️  TO JS undefined
⚡️  To Native ->  BluetoothLe getServices 104953974
⚡️  To Native ->  Preferences set 104953975
⚡️  TO JS {"services":[{"characteristics":[{"properties":{"writeWithoutResponse":false,"write":false,"extendedProperties":false,"read":false,"notifyEncryptionRequired":false,"authenticatedSignedWrites":false,"indicateEncryptionRequired":false,"indicate":false,"broad
⚡️  TO JS undefined
⚡️  To Native ->  BluetoothLe getServices 104953976
⚡️  TO JS {"services":[{"characteristics":[{"properties":{"writeWithoutResponse":false,"write":false,"extendedProperties":false,"read":false,"notifyEncryptionRequired":false,"authenticatedSignedWrites":false,"indicateEncryptionRequired":false,"indicate":false,"broad
⚡️  To Native ->  BluetoothLe write 104953977
⚡️  BluetoothLe - Reject write|0003CDD0-0000-1000-8000-00805F9B0131|0003CDD2-0000-1000-8000-00805F9B0131 Write timeout.
ERROR MESSAGE:  {"errorMessage":"Write timeout.","message":"Write timeout."}
⚡️  [error] - {"errorMessage":"Write timeout.","message":"Write timeout."}
⚡️  To Native ->  BluetoothLe write 104953978
⚡️  [error] - ERROR Error: Uncaught (in promise): overlay does not exist
⚡️  BluetoothLe - Reject write|0003CDD0-0000-1000-8000-00805F9B0131|0003CDD2-0000-1000-8000-00805F9B0131 Write timeout.
ERROR MESSAGE:  {"errorMessage":"Write timeout.","message":"Write timeout."}
⚡️  [error] - {"errorMessage":"Write timeout.","message":"Write timeout."}
⚡️  [error] - ERROR Error: Uncaught (in promise): overlay does not exist

```

Here when fails:

```
⚡️  To Native ->  BluetoothLe initialize 104953982
2023-01-27 14:49:39.383132+0000 App[878:33659] [CoreBluetooth] XPC connection invalid
⚡️  BluetoothLe - Resolve initialize BLE powered on
⚡️  TO JS undefined
⚡️  To Native ->  BluetoothLe isEnabled 104953983
⚡️  TO JS {"value":true}
⚡️  To Native ->  BluetoothLe getConnectedDevices 104953984
⚡️  TO JS {"devices":[]}
⚡️  To Native ->  BluetoothLe addListener 104953985
⚡️  To Native ->  BluetoothLe requestLEScan 104953986
⚡️  BluetoothLe - Resolve startScanning Scan started.
⚡️  TO JS undefined
⚡️  BluetoothLe - New device found:  GLAMM
⚡️  TO JS {"device":{"name":"GLAMM","deviceId":"FEE99B3A-8E41-43CB-7EAF-8B1B94A46EBC"},"rssi":-81,"txPower":127,"uuids":[],"localName":"GLAMM"}
⚡️  To Native ->  BluetoothLe removeListener 104953987
⚡️  To Native ->  BluetoothLe stopLEScan 104953988
⚡️  BluetoothLe - Stop scanning.
⚡️  TO JS undefined
⚡️  To Native ->  BluetoothLe removeListener 104953989
⚡️  To Native ->  BluetoothLe addListener 104953990
⚡️  To Native ->  BluetoothLe connect 104953991
⚡️  BluetoothLe - Connecting to peripheral <CBPeripheral: 0x283a200d0, identifier = FEE99B3A-8E41-43CB-7EAF-8B1B94A46EBC, name = GLAMM, mtu = 0, state = disconnected>
⚡️  BluetoothLe - Reject connect|FEE99B3A-8E41-43CB-7EAF-8B1B94A46EBC Failed to encrypt the connection, the connection has timed out unexpectedly.
ERROR MESSAGE:  {"message":"Failed to encrypt the connection, the connection has timed out unexpectedly.","errorMessage":"Failed to encrypt the connection, the connection has timed out unexpectedly."}
⚡️  [error] - {"message":"Failed to encrypt the connection, the connection has timed out unexpectedly.","errorMessage":"Failed to encrypt the connection, the connection has timed out unexpectedly."}
⚡️  BluetoothLe - Reject connect Connection timeout
ERROR MESSAGE:  {"message":"Connection timeout","errorMessage":"Connection timeout"}
⚡️  [error] - {"message":"Connection timeout","errorMessage":"Connection timeout"}
⚡️  [warn] - {"message":"Connection timeout","errorMessage":"Connection timeout"}

```

I don't know how to solve this.

pwespi, 2023-01-29:
Thanks for the logs and the link to the stackoverflow post. After reading some more about this, it seems that it is an issue with iOS. So I don't think there is anything that can be done in the pluin.

closed 2023-03-11 by marcelowolfsmartindustries