capacitor-community/bluetooth-le #487: Detecting if nearby devices permission is enabled

tylerclark, 2023-01-30:
**Is your feature request related to a problem? Please describe.**
I was having trouble trying to figure out why my iOS app worked and my Android app did not. Turns out at some point the Android OS revoked the permission for "Nearby Devices" since there was a checkbox in the Settings > App called "Remove Permissions if app is unused".

**Describe the solution you'd like**
Is there a way to detect that the permission for Nearby devices has been removed?

**Describe alternatives you've considered**
Any other way to detect this?

pwespi, 2023-02-27:
Thank you for this feature request. Currently `checkPermissions` is not implemented in this plugin, but it should be. I will take a look (but I'm very busy currently).

tylerclark, 2023-02-27:
Thanks Patrick! Would love some sort of event listener so if somebody turns off bluetooth the UI does something or they turn on bluetooth something happens (probably more so this use case). Not sure if that is possible.

pwespi, 2023-02-27:
> Thanks Patrick! Would love some sort of event listener so if somebody turns off bluetooth the UI does something or they turn on bluetooth something happens (probably more so this use case). Not sure if that is possible.

For this there is alreay https://github.com/capacitor-community/bluetooth-le#startenablednotifications

pwespi, 2023-04-02:
Closing this as duplicate of #445

closed 2023-04-02 by pwespi