capacitor-community/bluetooth-le #505: Support bluetooth-le for Electron platform

gregoiregentil, 2023-03-04:
Capacitor community bluetooth-le supports web platform (if web bluetooth is supported by the browser).

If the plugin is run from a capacitor Electron app, the following error message appears:
"scanForBluetoothDevices Error  Error: Web Bluetooth API not available in this browser."

It would be great to support Electron apps.

I do think that the web code "src/web.ts" could be re-used as Electron is leveraging chromium browser which supports web bluetooth. More info is available at the top of this page: https://www.electronjs.org/docs/latest/tutorial/devices

gregoiregentil, 2023-03-07:
I have managed to make it work by hacking but capacitor electron is so messy that I don't know how it could be integrated properly.

pwespi, 2023-03-07:
> I have managed to make it work by hacking but capacitor electron is so messy that I don't know how it could be integrated properly.

Nice that you have managed to make it work. So you added code to handle the `select-bluetooth-device` event and managed to reuse `src/web.ts`? 

Unfortunately I don't have any experience with capacitor electron.

gregoiregentil, 2023-03-07:
That is correct. select-bluetooth-device needs to be implemented.

I find capacitor Electron 'complicated'. I really don't know how a capacitor plugin could properly implement some Electron code especially in this case which you need to do some back and forth between Electron, plugin and main app... What I have done is to hack my own project. Anyway, this issue is labeled "enhancement", not "bug".

jrobeson, 2023-03-30:
another option if you want full support is to use a native module like noble instead. If it's still being maintained, then last i heard it could use the native bluetooth on the system, rather than the more limited webbluetooth api.

gregoiregentil, 2023-03-30:
Thank you for the good suggestion. The integration still remains complicated because of the overall architecture has so many IP blocks (Ionic, Capacitor, Electron, then noble). I'm fine with my ugly hack for the moment. I think that this issue is a kind of long-term enhancement.

hugocxl, 2024-06-10:
Hi @gregoiregentil, any chance that you could share your solution? I'm having the same issue here, thanks in advance! 🙏

gregoiregentil, 2024-06-10:
I don't know exactly how I have managed to make this work. My code is too messy to share but I see those two pieces of code that are definitely necessary:

```
app.commandLine.appendSwitch(os.platform() == "linux" ? "enable-experimental-web-platform-features" : "enable-web-bluetooth", "true");
```

and also:

```
	this.MainWindow.webContents.on('select-bluetooth-device', (event, devices, callback) => {
		event.preventDefault();
		console.log("Device list: ", devices);
		//insert your logic here to deal with the list of detected devices and call the function "callback" with the deviceID of the device that you want to use
    });
```

hugocxl, 2024-06-10:
@gregoiregentil sure that helps! Thanks ❤️