capacitor-community/bluetooth-le #508: Disconnect Lies Sometimes

vicatcu, 2023-03-10:
**Describe the bug**
**Sometimes** (I want to stress this is an intermittent problem, difficult to reproduce deterministically) the `await BleClient.disconnect(deviceId)` succeeds (doesn't throw an error) but the phone doesn't actually disconnect from the Bluetooth device. The onDisconnect callback attached to the `BleClient.connect` even fires after the `BleClient.disconnect` call, even though truly the phone is still connected to the Bluetooth device. I've tried, for example, repeatedly calling disconnect 3 times with 0.5s between attempts, to no avail. The only workaround is to turn Bluetooth off on the phone, which then positively results in the Bluetooth device indicating it has been disconnected from.

**To Reproduce**
Sorry, I can't provide predictable steps to reproduce this, but it happens "often enough" that I notice it and I think I'm going to have to recommend to my users that they turn off Bluetooth after finishing the configuration process that takes place over Bluetooth that I'm guiding them through in my app as a precaution against the device being left in configuration mode and chewing up batteries.

**Expected behavior**
`BleClient.disconnect(deviceId)` should reliably result in the phone no longer being connected to the Buetooth device and should not fire the onDisconnect callback associated with the associated `BleClient.connect` until the Bluetooth device is actually disconnected. It should certainly not resolve the Promise successfully if disconnect didn't actually happen.

**Screenshots**
N/A

**Plugin version:**
- @capacitor-community/bluetooth-le: **2.0.0**

**Desktop (please complete the following information):**
n/a

**Smartphone (please complete the following information):**
- Device: Google Pixel 6
- OS: Android 13
- Ionic App
- @ionic/angular: "^6.2.7", @capacitor/core: "^4.2.0"

**Additional context**
N/A

pwespi, 2023-03-12:
When the disconnect call resolves but the device stays connected, do you see the "Disconnected from GATT server" log statement in Android studio? If possible, can you share the logs from a case where the device disconnects and one where it does not?

Does the device still show up if you call `getConnectedDevices`?

Are you able to reproduce it with different Android phones?

vicatcu, 2023-03-13:
@pwespi  I can confirm that I see `D/Device: Disconnected from GATT server.` in logcat in Android Studio under these circumstances, despite my device beeping at me periodically to tell me it still believes it is connected. Turning Bluetooth on the phone off causes the device to disconnect / recognize it is disconnected an it indicate that to me audibly.

I can also confirm that 250ms, 500ms, and 750ms after `disconnect` is called if I call `getConnectedDevices`, my device still appears in the list (even after the `onDisconnect` callback fires). This seems to be true even when disconnect genuinely succeeds. As an aside, it doesn't seem like my `services` array parameter to `getConnectedDevices` actually filters _anything_ out of the results. I don't have a bunch of phones to test with unfortunately.

pwespi, 2023-03-13:
If the log statement "Disconnected from GATT server" appears, this means that the Android Bluetooth API reported that the device is now in `STATE_DISCONNECTED`, see https://github.com/capacitor-community/bluetooth-le/blob/bb3fe18939d71164ea89e641175393ba724de52c/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt#L54-L60

Therefore I'm not sure if there is anything that can be done about this in the plugin. 

Is there another app installed that connects to the peripheral? Or was the initial connection established in the Android settings instead of the app? Is there bonding involved?

vicatcu, 2023-03-13:
Interesting... I have nRF connect installed on the phone, but it's not open. Could it be running in the background or something and interfering somehow I wonder?

pwespi, 2023-04-15:
I found the problem. This happened when `connect` was called multiple times. In that case the plugin created multiple internal GATT clients, but in the disconnect call only one of the clients was disconnected.

Fixed in 2.2.3.

closed 2023-04-15 by pwespi

aanders77, 2024-01-16:
I'm on version 3.1.1, but I'm experiencing the same issue. Below is the log from Chrome Devtools. What happens here is the following:
1. I'm attempting a write operation, which fails with error.message "Not connected to device."
2. I'm running getConnectedDevices(), and it claims that the device is still connected (MAC starting with C0:EA)
3. I'm running stopNotifications(), which also fails because "Not connected to device."
4. Because of this, I'm running disconnect(). This call seems to be successful, no error is being catched.
5. I'm running getConnectedDevices() again, and the device still shows up being connected.
[image]