capacitor-community/bluetooth-le #510: Caused by java.lang.IllegalStateException BT Adapter is not turned ON

lshaoshuai, 2023-03-22:
any android app

**Turn off bluetooth when i scan for bluetooth devices，I get the following exception and the app crashes.**

**Caused by java.lang.IllegalStateException
BT Adapter is not turned ON**

android.bluetooth.le.BluetoothLeUtils.checkAdapterStateOn (BluetoothLeUtils.java:191)
android.bluetooth.le.BluetoothLeScanner.stopScan (BluetoothLeScanner.java:318)
com.capacitorjs.community.plugins.bluetoothle.DeviceScanner.stopScanning (DeviceScanner.kt:134)
com.capacitorjs.community.plugins.bluetoothle.BluetoothLe.requestLEScan (BluetoothLe.kt:316)
java.lang.reflect.Method.invoke (Method.java)
com.getcapacitor.PluginHandle.invoke (PluginHandle.java:138)
com.getcapacitor.Bridge.lambda$callPluginMethod$0$com-getcapacitor-Bridge (Bridge.java:763)
com.getcapacitor.Bridge$$ExternalSyntheticLambda5.run (未知来源:8)
android.os.Handler.handleCallback (Handler.java:942)
android.os.Handler.dispatchMessage (Handler.java:99)
android.os.Looper.loopOnce (Looper.java:240)
android.os.Looper.loop (Looper.java:351)
android.os.HandlerThread.run (HandlerThread.java:67)

pwespi, 2023-03-28:
I cannot reproduce this. Which version of Android and which version of the plugin is this?

pwespi, 2023-04-02:
Thank you for reporting this. Version 2.2.0 contains a fix for this. Note that I wasn't able to fully test the fix, because I wasn't able to reproduce the error.

closed 2023-04-02 by pwespi

pwespi, 2023-04-02:
I assume this is related to #317, which also only happened in certain Android devices but not in others.