capacitor-community/bluetooth-le #512: iOS app crashes on Bluetooth Write

vicatcu, 2023-03-23:
**Describe the bug**
A clear and concise description of what the bug is. I could use some help getting diagnostic info from app running on iPhone. I've done my development using Android and it's working adequately on that platform.

**To Reproduce**
1. Install the App via TestFlight on an iOS phone
2. Successfully connect to a device after a Bluetooth Scan.
3. Try and write to a valid characteristic.
4. Observe app crashes.

**Expected behavior**
App should write the characteristic and return without crashing.

**Screenshots**
N/A

**Plugin version:**
- @capacitor-community/bluetooth-le: 2.1.0

**Smartphone (please complete the following information):**
- Device: iPhone 12
- OS: iOS 16.2
- Browser Ionic 6.2.7
- Version N/A

vicatcu, 2023-03-23:
Adding a crash report screenshot from TestFlight
[image]

vicatcu, 2023-03-23:
Here's what I'm seeing in XCode with an iPhone 11 connected to my computer:

```
To Native ->  BluetoothLe getConnectedDevices 62163661
2023-03-23 16:54:57.896491-0400 App[402:28704] *** Assertion failure in -[CBUUID initWithString:], CBUUID.m:169
2023-03-23 16:54:57.899823-0400 App[402:28704] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'String  does not represent a valid UUID'
*** First throw call stack:
(0x19f1ed86c 0x1b4206c50 0x19f0f3000 0x1a048791c 0x1b8e5696c 0x1b8e56c64 0x10463c1a4 0x10463c150 0x1a2f86128 0x10463bdd8 0x10463c308 0x10447273c 0x1044668a0 0x104803bcc 0x1048056c0 0x10480d354 0x10480e0c0 0x10481a644 0x1eac99814 0x1eaca076c)
libc++abi.dylib: terminating with uncaught exception of type NSException
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'String  does not represent a valid UUID'
terminating with uncaught exception of type NSException
```

vicatcu, 2023-03-23:
OK it turns out this is what happens when you do `BleClient.getConnectedDevices(['']);`

pwespi, 2023-04-02:
Thanks for reporting and debugging this. In version 2.2.0 calling `BleClient.getConnectedDevices([''])` will now reject the call in the JavaScript layer, instead of crashing the app.