capacitor-community/bluetooth-le #516: Descriptor 0x2901 "Characteristic User Description" is being read differently across platforms

Arxi, 2023-04-02:
**Describe the bug**
It seems that descriptor with UUID `0x2901` (`00002901-0000-1000-8000-00805f9b34fb`), also known as "Characteristic User Description", has special significance and is being read differently on Web, Android and iOS.
If we try to use `readDescriptor` on two descriptors with identical values, one of them having UUID `0x2901`, and the other any other random UUID (e.g. `0x29f2`), we'll see different byte lengths in the resulting ArrayBuffers.

**To Reproduce**
First, we need a peripheral which contains a services with a characteristic which has "Characteristic User Description" descriptor. Such peripheral can be simulated by using nRF Connect app for Android, or EFR Connect app for iOS. Here's the configuration of a simulated peripheral that can be loaded into nRF Connect app:

```xml
<server-configuration name="desc_test">
   <service name="service1" uuid="00001234-0000-1000-8000-00805f9b34fb">
      <characteristic name="characteristic1" uuid="00009999-0000-1000-8000-00805f9b34fb" value-string="hello">
         <descriptor uuid="00002901-0000-1000-8000-00805f9b34fb" value-string="abcdefgh">
            <permission name="READ"/>
         </descriptor>
         <descriptor name="descriptor2" uuid="000029f2-0000-1000-8000-00805f9b34fb" value-string="abcdefgh">
            <permission name="READ"/>
         </descriptor>
         <property name="READ"/>
         <property name="WRITE"/>
      </characteristic>
   </service>
</server-configuration>
```

There's just one service with id `00001234-0000-1000-8000-00805f9b34fb`, characteristic `00009999-0000-1000-8000-00805f9b34fb` and two descriptors:
`00002901-0000-1000-8000-00805f9b34fb` (also known as "Characteristic User Description")
and
`000029f2-0000-1000-8000-00805f9b34fb` which is a random custom descriptor I chose.
Both descriptors have the same value: string `abcdefgh`.

We can use the following piece of code to test how both of these descriptors are being read. The code requests the peripheral (simulated from nRF Connect app on some Android phone, for example), connects to it, then reads both descriptors from the single characteristic. The byte lengths of both readings should be 8 (the character length of string `abcdefgh`)

```js
  const descriptorsTest = async () => {
    try {
      console.log("Running descriptorsTest");

      await BleClient.initialize({androidNeverForLocation: true});
      const peripheral = await BleClient.requestDevice({
        optionalServices: ["00001234-0000-1000-8000-00805f9b34fb"]
      });
      await BleClient.connect(
        peripheral.deviceId,
        (id) => console.log(`Device disconnected: ${id}`),
        {timeout: 15000}
      );

      // read the "Characteristic User Description" descriptor (0x2902)
      const characteristicUserDescriptionValue = await BleClient.readDescriptor(
        peripheral.deviceId,
        "00001234-0000-1000-8000-00805f9b34fb",
        "00009999-0000-1000-8000-00805f9b34fb",
        "00002901-0000-1000-8000-00805f9b34fb"
      );

      // read the other descriptor (0x29f2)
      const differentDescriptorValue = await BleClient.readDescriptor(
        peripheral.deviceId,
        "00001234-0000-1000-8000-00805f9b34fb",
        "00009999-0000-1000-8000-00805f9b34fb",
        "000029f2-0000-1000-8000-00805f9b34fb"
      );

      // the byte lengths of both results should be 8
      console.log(`0x2901 read result byte length: ${characteristicUserDescriptionValue.buffer.byteLength}`);
      console.log(`0x29f2 read result byte length: ${differentDescriptorValue.buffer.byteLength}`);

    } catch (error) {
      console.log("Error!");
      console.log(error);
    }
  }
```

When we run this code on **Web** platform (Chrome on MacOS), we see this in console output:

```
0x2901 read result byte length: 8
0x29f2 read result byte length: 8
```

On **Android**, we see this: 
```
0x2901 read result byte length: 9
0x29f2 read result byte length: 8
```
Which means the value of descriptor `0x2901` now contains one extra character at the end.

On **iOS**, the result is the weirdest:
```
0x2901 read result byte length: 1
0x29f2 read result byte length: 8
```
But BluetoothLe logs in XCode give us a hint about what's happening here:
```
⚡️  BluetoothLe - Resolve readDescriptor|00001234-0000-1000-8000-00805F9B34FB|00009999-0000-1000-8000-00805F9B34FB|00002901-0000-1000-8000-00805F9B34FB abcdefgh
⚡️  TO JS {"value":"abcdefgh"}
⚡️  BluetoothLe - Resolve readDescriptor|00001234-0000-1000-8000-00805F9B34FB|00009999-0000-1000-8000-00805F9B34FB|000029F2-0000-1000-8000-00805F9B34FB 61 62 63 64 65 66 67 68 
⚡️  TO JS {"value":"61 62 63 64 65 66 67 68 "}
```
It seems the descriptor `0x2901` is being directly translated to the UTF8 string (by the platform?), and my assumption is that bluetooth-le plugin isn't expecting this. Unfortunately, I'm not knowledgeable in native iOS code, so please take that with a grain of salt.

**Expected behavior**
The result of using `readDescriptor()` on a descriptor with UUID `0x2901` should be the same on all platforms (in this case, 8 bytes).

**Plugin version:**
- @capacitor-community/bluetooth-le: 2.2.0

**Desktop**
- OS: MacOS Ventura 13.2.1
- Browser: Chrome
- Version: 111.0.5563.146

**Smartphone**
- iOS device: iPhone7, iOS 15.4.1
- Android device: Xiaomi Redmi Note 8 Pro, MIUI Global 12.5.7. (Android 11)

**Additional context**
The question arises - are there any other descriptors which have special significance and are being read differently across platforms?

pwespi, 2023-04-08:
Thank you for reporting this.

You are right, iOS treats the 0x2901 descriptor specially and returns directly a UTF8 string, which the plugin didn't handle correctly. This is fixed in 2.2.1.

However, I'm not able to reproduce the Android behavior you're describing. In version 2.2.1 I'm now getting the same value on all platforms. 

Can you check what the values of the 9 bytes are that you are getting on Android compared to the other platforms?

pwespi, 2023-04-08:
> The question arises - are there any other descriptors which have special significance and are being read differently across platforms?

According to the [Assigned Numbers](https://www.bluetooth.com/specifications/assigned-numbers/) document, these are the descriptors with special significance:

```
0x2900 Characteristic Extended Properties
0x2901 Characteristic User Description
0x2902 Client Characteristic Configuration
0x2903 Server Characteristic Configuration
0x2904 Characteristic Presentation Format
0x2905 Characteristic Aggregate Format
0x2906 Valid Range
0x2907 External Report Reference
0x2908 Report Reference
0x2909 Number of Digitals
0x290A Value Trigger Setting
0x290B Environmental Sensing Configuration
0x290C Environmental Sensing Measurement
0x290D Environmental Sensing Trigger Setting
0x290E Time Trigger Setting
0x290F Complete BREDR Transport Block Data
```

If any other of them directly return a string on iOS, this should already be handled correctly now, but I haven't tested all of them.

Arxi, 2023-04-09:
Thank you so much for dealing with this so quickly @pwespi ! I just tried 2.2.1 and can confirm the iOS problem is fixed for `0x2901` descriptor. I haven't tried other descriptors yet.

As for Android problem - now, I have to admit, I have to test Android devices differently, because I have only one Android device. Therefore, when I want to read descriptors from my Android device, I can't use nRF Connect app on the same device as a peripheral simulator. 

So I'm using EFR Connect on my iPhone, where I manually created the same simulated peripheral with two descriptors, and I read those from Android phone and from Web. When I do that, I still see that resulting buffer's byte length for `0x2901` is 8 on Web and 9 on Android.

The last byte seems to simply be `00` (screenshot from console from chrome inspect connected to Android device):
[image]

But reading the same descriptor on the Web looks like this:
[image]

But maybe there's something I don't understand - if you have some other suggestions I could test or try, I'm happy to help! Otherwise, for my purposes, I consider this bug to be fixed, so thank you for that!

pwespi, 2023-04-15:
As the iOS part is fixed and I cannot reproduce the Android problem, I will close this issue for now. If it cases any more problems, let me know and I can take another look.

closed 2023-04-15 by pwespi