capacitor-community/bluetooth-le #528: iOS 16.4.1 Scan doesn't return service uuids

eppleton, 2023-04-26:
Up to iOS version 16.4.1 a BLE Scan without setting a filter for service ids would return devices with the service uuids from the advertisement. This behavior has changed and in version 16.4.1 the uuids are no longer returned. 

Applying a filter for service uuids fixes thew problem. For background: we have a cross platform app and had to disable filters, because some older Android devices do not support filtering for services (they will simply not find any devices).

ASHBAW, 2023-04-26:
It may be related to my #527 issue (missing local name in scan response). Somebody made the suggestion of trying allowDuplicates=true in the scan options. You will get duplicates but I am now able to see the local name.

eppleton, 2023-04-26:
Thanks for the quick reply, I can confirm that I do have the issue with local name as well, but it's not a problem in my case. I fixed my problem by applying the filter. 

One of the problems with these issues is, that they don't occur on all devices. On one of my devices, an iPhone 11 Pro Max that I usually use for testing the issue doesn't appear. I think the reason might be that Apple caches some information about devices like the device name and maybe also service uuids. If you have scanned the device before the upgrade to 16.4.1 the information is in the cache and everything seems to work, whereas new users of the app would experience the issue.

I was only able to reproduce the issue after I used an iPhone that didn't have our app installed before.

eppleton, 2023-04-26:
I suspect this and the local name issue is not a bug in bluetooth-le, but rather in iOS bluetooth core: https://developer.apple.com/forums/thread/728842

peitschie, 2023-04-27:
@eppleton iOS caches a lot of data about previously seen Bluetooth devices. To fully clear the caches, I believe you need a full phone factory reset to avoid the caching behaviour there for advertising packets.

LgArcida, 2023-12-20:
Still an issue, is there any workaround on this?

ASHBAW, 2023-12-20:
Somebody made the suggestion of trying allowDuplicates=true in the scan options. You will get duplicates but I am now able to see the local name.

LgArcida, 2023-12-20:
Yeah tried that one but still the same
Name is partially correct and when adding any other type of filter like optionalServices devices wont be visible anymore

robsonos, 2024-02-08:
Hi,
I am facing a similar issue. The number of UUIDs returned from iOS is smaller than from Android.
Is there any progress on this? Is there anything I can help with?