capacitor-community/bluetooth-le #529: Upgrade Java compatibility from 11 to 17

y3nd, 2023-05-04:
**Describe the bug**
Unable to compile on Android since all modules in Capacitor 5.x are built with Java 17 and this module is compiled with Java 11 in accordance to build.gradle file

**To Reproduce**
Build an app with capacitor 5.x

**Expected behavior**
Should compile with Java 17 but build.gradle is blocking compilation
Upgrading to 17 like [other plugins](https://github.com/ionic-team/capacitor-plugins/blob/74f64fa1992a89519aa22e310cb4060c758acdeb/text-zoom/android/build.gradle#L52) in the [build.gradle file works but needs testing](https://github.com/capacitor-community/bluetooth-le/blob/b421d2937ff840e71723d19ceef0dc479f599a2d/android/build.gradle#L45)

**Plugin version:**
- @capacitor-community/bluetooth-le: main branch

Arxi, 2023-05-07:
I have this problem as well and can confirm changing the `sourceCompatibility` and `targetCompatibility` in the plugin's build.gradle file to `JavaVersion.VERSION_17` seems to help in my case too.

jrobeson, 2023-05-07:
If you folks know it works, why not submit a PR.

closed 2023-05-07 by pwespi