capacitor-community/bluetooth-le #531: Error installing @bluetooth-le@3.0.0-1

unstopp, 2023-05-04:
Hello,

For some reason I am unable to install the package. I get the following error:

**Invalid response body while trying to fetch https://registry.npmjs.org/@types%2fweb-bluetooth: ENOENT: no such file or directory, rename '/Users/marius/.npm/_cacache/tmp/d2560a15' -> '/Users/marius/.npm/_cacache/content-v2/sha512/2d/69/90b45908e84a5b46345d66597ff6ea81bd2be37fa362afd48e09f101c3882a79042ba6e35c0412dbb198648549aec2dc147f729053cbf7e45f73c6cc0061'**

Renaming doesn't help, that file does not exist. Any help is very much appreciated!

Thank you!
Marius

Arxi, 2023-05-06:
What is the command you are using to install version 3.0.0-1 of the package? I was successful using this:
`npm install @capacitor-community/bluetooth-le@next`

unstopp, 2023-05-07:
> What is the command you are using to install version 3.0.0-1 of the package? I was successful using this: `npm install @capacitor-community/bluetooth-le@next`

Thank you for your reply. I am getting the same error..

Arxi, 2023-05-07:
Sorry to hear it didn't help! I'm not an expert on NPM, but this seems more like an npm or npm cache problem than a problem with this specific package. 
Are you able to install other packages or other versions of this plugin?
What npm and node version are you using?
What operating system?

I'd try some of the ideas here (cleaning the cache, reinstalling NPM, or using NVM to install other or newer version of node and try again): 
https://stackoverflow.com/questions/72856997/npm-invalid-response-body-while-trying-to-fetch-npm-cant-install-any-depende

unstopp, 2023-05-07:
> Sorry to hear it didn't help! I'm not an expert on NPM (nor this package), but this seems more like an npm or npm cache problem than a problem with this specific package. Are you able to install other packages or other versions of this plugin? What npm and node version are you using? What operating system?
> 
> I'd try some of the ideas here (cleaning the cache, reinstalling NPM, or using NVM to install other or newer version of node and try again): https://stackoverflow.com/questions/72856997/npm-invalid-response-body-while-trying-to-fetch-npm-cant-install-any-depende

I tried clearing cache in the past but not with "npm cache clean --force"
This fixed the problem and installed successfully. Thank you for pointing it out!

closed 2023-05-07 by unstopp