capacitor-community/bluetooth-le #542: BLE Web discovery not working

Alvaro948, 2023-05-18:
**Describe the bug**
On Chrome 113, the discovery returns a promise thhat never resolves

**To Reproduce**
```ts
await BleClient.requestLEScan(
    {
        namePrefix: 'MY_PREFIX'
    },
    (result: ScanResult) => {
        console.log(result)
    }
);
```

**Plugin version:**
- @capacitor-community/bluetooth-le: 3.0.0

**Desktop (please complete the following information):**
- OS: Windows and Linux at least
- Browser Chrome at least
- Version 113.0.5672.126

**Smartphone (please complete the following information):**
No affected

**Additional context**
As mentioned here https://github.com/reelyactive/web-ble-demo/issues/2, the usage of `requestLEScan` seems to be deprecated in favour of `requestDevice`

pwespi, 2023-05-18:
Thank you for the issue.

Unfortunately, `requestLEScan` is still an experimental feature and has never really been stable in Chrome (nor any other browser).

See also https://bugs.chromium.org/p/chromium/issues/list?q=requestLEScan

I'm not sure if there is anything that can be done about it in this plugin. If you're targeting the web platform, I would suggest to use `requestDevice`.

Alvaro948, 2023-05-18:
Hi @pwespi, thank you for the quick reply.

Yes, maybe I  have overthink about this, and the docs indeed indicate that `requestLEScan` is unstable for web

closed 2023-05-18 by Alvaro948