capacitor-community/bluetooth-le #551: Deprecation notices

muuvmuuv, 2023-07-05:
**Describe the bug**

Latest Android Studios reports some deprecations.

```
> Task :capacitor-community-bluetooth-le:compileDebugKotlin
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/BluetoothLe.kt: (153, 40): 'enable(): Boolean' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/BluetoothLe.kt: (164, 40): 'disable(): Boolean' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (99, 22): This declaration overrides deprecated member but not marked as deprecated itself. This deprecation won't be inherited in future releases. Please add @Deprecated annotation or suppress
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (102, 19): 'onCharacteristicRead(BluetoothGatt!, BluetoothGattCharacteristic!, Int): Unit' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (105, 43): 'getter for value: ByteArray!' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (130, 22): This declaration overrides deprecated member but not marked as deprecated itself. This deprecation won't be inherited in future releases. Please add @Deprecated annotation or suppress
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (133, 19): 'onCharacteristicChanged(BluetoothGatt!, BluetoothGattCharacteristic!): Unit' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (135, 39): 'getter for value: ByteArray!' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (142, 22): This declaration overrides deprecated member but not marked as deprecated itself. This deprecation won't be inherited in future releases. Please add @Deprecated annotation or suppress
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (145, 19): 'onDescriptorRead(BluetoothGatt!, BluetoothGattDescriptor!, Int): Unit' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (149, 39): 'getter for value: ByteArray!' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (263, 36): 'getParcelableExtra(String!): T?' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (391, 24): 'setter for value: ByteArray!' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (393, 37): 'writeCharacteristic(BluetoothGattCharacteristic!): Boolean' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (435, 28): 'setter for value: ByteArray!' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (437, 28): 'setter for value: ByteArray!' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (440, 24): 'setter for value: ByteArray!' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (443, 41): 'writeDescriptor(BluetoothGattDescriptor!): Boolean' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (501, 20): 'setter for value: ByteArray!' is deprecated. Deprecated in Java
w: //@capacitor-community/bluetooth-le/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt: (502, 37): 'writeDescriptor(BluetoothGattDescriptor!): Boolean' is deprecated. Deprecated in Java
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Plugin version:**
- @capacitor-community/bluetooth-le: [e.g. 0.4.0]

**Desktop (please complete the following information):**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Smartphone (please complete the following information):**
- Device: [e.g. iPhone6]
- OS: [e.g. iOS8.1]
- Browser [e.g. stock browser, safari]
- Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

pwespi, 2023-07-10:
This plugin uses the same versions of Java, Kotlin, minSdkVersion, etc. as Capacitor core and the official Capacitor plugins. So for now these warnings shouldn't cause any problems and are just warnings.

For the future the code will need to be updated. I will keep this issue to track this.

closed 2023-10-03 by pwespi

pwespi, 2023-10-03:
We have changed the code to use the new Android methods from API level 33 for Android >= 13. However the old code is still there as well to support older devices. Therefore the deprecation warnings will still appear until we can remove the old code, which will take some time.

PR #572 (2023-10-03): Android api 33 methods
Partially resolves #551