capacitor-community/bluetooth-le #552: BLE not available in iOS simulator

sarink, 2023-07-08:
**Describe the bug**
Can't seem to get this to work in iOS, not sure what steps to take to begin debugging? I can't get the lib to work at all

**To Reproduce**
I have the following simple react component:
```
import { BleClient } from '@capacitor-community/bluetooth-le';
import { Bluetooth } from '@mui/icons-material';
import React from 'react';

export const DashboardPage= () => {
  return (
    <div>
      <Button
        variant="contained"
        startIcon={<Bluetooth />}
        onClick={() => {
          BleClient.initialize()
            .then(() => {
              alert('BLE available');
            })
            .catch(() => {
              alert('BLE not available');
            });
        }}
      >
        Scan for bluetooth devices
      </Button>
    </div>
  );
};
```

**Expected behavior**
When clicking the button, it alerts that ble is not available

**Plugin version:**
`^3.0.0`

**Smartphone (please complete the following information):**
iPhone 14 simulator

**Additional context**
Add any other context about the problem here.

pwespi, 2023-07-10:
> Smartphone (please complete the following information):
iPhone 14 simulator

Bluetooth is not available in the iOS simulator. You have to test with a real device.

closed 2023-07-10 by pwespi

sarink, 2023-07-11:
Oh, didn't realize that. Might be nice to mention it in the docs. Thanks!

pwespi, 2023-07-11:
Good idea. Thanks!