capacitor-community/bluetooth-le #557: Notification received whilst being disconnected.

sanastasiou, 2023-07-30:
**Describe the bug**
Connect to BLE device and start notification.

Restart BLE device to break connection and observe that notification is still being received, however attemtp to e.g. call readRssi leads to Not connected to device exception.

```
                await BleClient.startNotifications(device.deviceId, this.KILLSWITCH_SID, this.PING_CHARACTERISTIC,
                    async (value) => {
                        const counterValue = value.getUint32(0, true);
                        console.log(`Killswitch counter value: ${counterValue} - UUID: ${this.PING_CHARACTERISTIC}`);
                        //refresh timer
                        this.resetDeviceTimeout(scanResult.device.deviceId);
                        const rssi = await BleClient.readRssi(device.deviceId);
                        this.fKillSwitchStates[device.deviceId].rssi       = this.rssiToClassConverter(rssi);
                        this.fKillSwitchStates[device.deviceId].rssi_value = rssi;
                        if(this.fChangeRef != null)
                        {
                            this.fChangeRef.detectChanges();
                        }
                    });
```

I don't understand how above code leads to:

[image]

Since the device was disconnected, why is the notification still being received? Do I have to call something during the disconnect callback?

pwespi, 2023-08-01:
What does the timing look like of restarting the BLE device and the error?
It could be that the last notification is received and just in that moment you restart the device and then the next call fails?

Does it happen all the time or just sometimes? Is this on Android or iOS? What plugin version?

sanastasiou, 2023-08-01:
Hi @pwespi I thought about that, but it happened repeatedly until it "reconnected".. Which makes no sense to me. I am using capacitor 3 so my plugin is pretty old I guess `"@capacitor-community/bluetooth-le": "^1.8.2",` .

sanastasiou, 2023-08-03:
I was not able to reproduce this thus far, with the most up to date plugin version. I guess we can consider this issue resolved.

pwespi, 2023-08-10:
Thanks for reporting back.

closed 2023-08-10 by pwespi