capacitor-community/bluetooth-le #567: Timeout on `createBond`

micheleypf, 2023-09-20:
**Is your feature request related to a problem? Please describe.**
Out in the wild there are BLE devices I have come across that do not generate the bonded event from Android. That causes the plugin to wait for such event forever.

**Describe the solution you'd like**
The solution is rather simple, add a timeout at the end of `Device#createBond`.

**Describe alternatives you've considered**
Another idea might be to receive the result via an event handler.

**Additional context**
If you find that this proposal makes sense, I'll be more than happy to make a PR

pwespi, 2023-10-01:
Thank you for this issue.

>Another idea might be to receive the result via an event handler.
What do you mean by this?

I think adding a timeout to `createBond` is a good idea, but you wouldn't know if the bonding was successful or not if you run into the timeout, right?

PRs are welcome.

micheleypf, 2023-10-01:
> I think adding a timeout to `createBond` is a good idea, but you wouldn't know if the bonding was successful or not if you run into the timeout, right?

Yeah you would not. But you would have the plugin freeze waiting for an event that will never fire. If something went wrong in the bonding process, the call would be rejected before listening for the bonded event. I also observed that in these cases Android simply assumes the periferal is bonded after a while, unfortunately I did not find any documentation peace to back up my claim.

> What do you mean by this?

Adding an `addListener` to the api specific for this. That would make the `createBond` resolve and let the user handle either the result or the timeout.

I'll try to make a PR tomorrow, thanks!

pwespi, 2023-10-02:
closed by https://github.com/capacitor-community/bluetooth-le/pull/573

closed 2023-10-02 by pwespi