capacitor-community/bluetooth-le #581: Not able to get the scanned data for Zebra barcode scanners

aditya-babu1, 2023-10-18:
**Describe the bug**
We are able to connect with the Zebra Barcode device(DS2278 model), but after scanning data transmission is not happening, does this plugin support all the devices in low energy mode?

**Plugin version:**
- @capacitor-community/bluetooth-le: 2.2.3

pwespi, 2023-10-18:
Yes, this plugin should support all Bluetooth Low Energy devices (however not Bluetooth classic).

Do you know the Bluetooth Low Energy protocol used by this Zebra device? You will have to use `startNotifications` with a specific service and characteristic to obtain any data.

aditya-babu1, 2023-10-18:
yes @pwespi, we tried using startNotifications with all service and characteristic ids having notify/indicate true individually, also tried enabling notifications using appropriate descriptor id and then used startNotifications with appropriate service and characteristic id, but data-transfer is not happening from zebra barcode scanner, do you see anything wrong here, can you please help on this?

pwespi, 2023-10-19:
Do you have the specification of the Zebra device? Maybe you need to write some configuration to a characteristic to enable the data transmission.