capacitor-community/bluetooth-le #598: Is not a but exactly, but a question

barart, 2023-11-20:
Is there a way to check if its already initialized? the isEnabled() is for check if bt is enabled on system, no if the component was already initialized, the initialize() function dont return anything to see if it was initialized or not

mutasimissa, 2023-11-20:
@barart Please remove the bug tag amd move this to discussion board.
wrap the `initialize()` with try/catch block and you can also use `isEnabled()` , I don't think there is another way in the native platforms itself to check for initialization
`  
try {
      await BleClient.initialize();
      let isBtEnabled = await BleClient.isEnabled();
      if (!isBtEnabled) {
        throw new Error("BLE not available")
      }
   } catch (e) {
      console.log(`${e}`);
   }
`

barart, 2023-11-20:
@mutasimissa THanks, will check, i cant change the label (i dont have permission)