capacitor-community/bluetooth-le #600: Android Connection timeout

Rossella-Mascia-Neosyn, 2023-11-23:
**Describe the bug**
In my app it often happens that it times out

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the device to the ble service
2. See error

**Expected behavior**
It should never return connect timeout

**My code**
```
  const scan = async (): Promise<void> => {
    try {
      await BleClient.initialize();
      const isEnable = await checkBluetoothStatus();
      if (isEnable) {
        const device = await BleClient.requestDevice({
          scanMode: ScanMode.SCAN_MODE_LOW_POWER,
          namePrefix: 'SMART_LOCK',
        });
        await connectBt(true, device);
        await BleClient.stopLEScan();
        console.info('stopped scanning');
      }
    } catch (error) {
      console.error(error);
    }
  };

 const connectBt = async (withState: boolean, deviceParam?: BleDevice) => {
    try {
      withState && setIsLoadingBle(true);
      const deviceStored = await getLocker();
      const device: LockerStorage = { macAddress: (deviceParam?.deviceId ?? '') || (deviceStored.macAddress ?? '') };
      if (!device.macAddress) throw new Error('macAddress not found');
      await BleClient.disconnect(device.macAddress);
      await BleClient.connect(device.macAddress);
      const services = await BleClient.getServices(device.macAddress);
      if (services[3]) {
        deviceParam?.deviceId &&
          setLocker({
            macAddress: deviceParam.deviceId,
            service: services[3],
          });
        withState && setIsLoadingBle(false);
        withState && setIsConnectBle(true);
      } else {
        console.error('service not found', services);
      }
    } catch (e) {
      console.error(e);
      setIsLoadingBle(false);
      setIsConnectBle(false);
      await toast({
        behavior: 'error',
        message: (e as ErrorBle).message,
        duration: 5000,
      });
    }
  };
```
**Screenshots Logcat**
[image]

**Plugin version:**
- @capacitor-community/bluetooth-le: 3.1.1

**Smartphone (please complete the following information):**
- Device: Samsung Galaxy S8
- Android version 9

**Additional context**
I have already tried, as shown in my code, to insert disconnect and connect together

mutasimissa, 2023-11-23:
I don’t think this is a bug, the timeout means failed of connecting to a certain deviceId, please try to make sure you are connecting to the correct deviceId and if you are sure then try connecting to other peripheral and see if the problem is the same.

Rossella-Mascia-Neosyn, 2023-11-24:
actually it's just a bug I tried with another device and I have the same problem.
I tried with: Redmi note 9 e android version 12.

I hope to be able to fix it together as soon as possible because for us this error is blocking

madhan-SEET, 2024-02-28:
I'm also facing the same issue (Connection Timeout) . I tried to disconnect before connect still Its not working .

quadratech-dev-admin, 2024-04-28:
Any solution for this?

madhan-SEET, 2024-05-02:
I just increase my ble device advertisement packets previously 1 sec a packet and now 200 mill sec a packet . Now the connection timeout error decreased . Its not coming frequently but I cant able to avoid totally .

peitschie, 2024-05-02:
The BLE advertising interval absolutely can impact this significantly. Even 200ms is still only middling. Apple recommends an interval of 20ms for at least the first 30s for best discovery, then has some suggested intervals to slow down at.

See https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf

[image]

Cypher103360, 2025-05-07:
I have two BLE devices with the same firmware version:

✅ Device 1:
**Connects instantly** — BluetoothGatt callbacks are smooth, with no delay in onConnectionStateChange or onServicesDiscovered.

**Sends data smoothly** — responds correctly to commands like BATTERY, characteristics are read/written successfully, and notifications work as expected.

❌ Device 2:
**Fails to connect reliably** — it either times out or gets stuck before onServicesDiscovered, and sometimes never reaches the connected state.

**Doesn’t respond to commands** — even if the connection is established, it doesn't reply to commands like BATTERY, VER, and characteristic reads return nothing.

So, my question is:

What could be the possible reason? Might there be an issue in the code or something wrong with the BLE device or the firmware version?

peitschie, 2025-05-09:
Hi @Cypher103360 

Do you see this same situation with another application (e.g., nrf Connect)?

Also, just to confirm, you're experiencing this issue on Android?

Cypher103360, 2025-05-09:
Hello @peitschie, Thanks a lot for your reply 😊

I tried connecting both devices to the **nRF Connect** application. Eventually, there was no issue with the connection; I was able to connect both devices instantly.

However, when I tried to communicate with the first device with commands like (BATTERY, VER, GETDATA), it replied in  **HEX**, but the second device didn't reply to even a single command.

| Device 1 | Device 2 |
|--------|-------|
| [image]| [image] |