capacitor-community/bluetooth-le #611: Android: failing to connect when native pair modal not pressed in time

simonvank98, 2024-01-08:
**Describe the bug**
Whenever I call the BleClient.Connect function, this will trigger a native modal on an android device where you need to press 'pair' or 'deny'. Whenever the user takes too long to press pair, the connect method will trigger its disconnected function and the rest of my flow fails. This is not what I expect, shouldn't the connect method have some type of waiting mechanism for this modal?

**To Reproduce**
1. Call BleClient.connect()
2. Wait a couple of seconds
3. Press pair

**Expected behavior**
I expect to succeed the connection, even when pair is not pressed inmediately

**Plugin version:**
- @capacitor-community/bluetooth-le: 3.1.0

**Desktop (please complete the following information):**
- OS: Android

**Smartphone (please complete the following information):**
- Device: One plus 10pro
- OS: Android version 14

pwespi, 2024-01-08:
You are right, currently the connect call does not wait for the pairing process. 

As a workaround you can set the timeout of the [connect](https://github.com/capacitor-community/bluetooth-le#connect) call to something longer than the default (10000 ms).

crebuh, 2024-02-13:
@pwespi tried the workaround but the promise directly resolves when the connection is established, so the pairing dialog always occurs afterwards. any other idea?