capacitor-community/bluetooth-le #628: Scanning possible when bluetooth is off

juniorsaraviao, 2024-02-26:
**Describe the bug**
When the Bluetooth is off, the plugin can still detect Bluetooth devices (this only happens on Android). Even it's possible to connect.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the project https://github.com/juniorsaraviao/BluetoothSample/tree/master
2. Run the project on Android.
3. Turn off the Bluetooth.
4. Tap Scan.

**Expected behavior**
The plugin shouldn't be able to detect devices when the Bluetooth is off.

**Screenshots**
| Issue |
| ------------- |
| <video src="https://github.com/juniorsaraviao/BluetoothSample/assets/43689290/d85523df-8cc6-4298-9da1-ba0db0b41e61" width=180 height=392> |

**Plugin version:**
- @capacitor-community/bluetooth-le: v3.1.2

**Desktop (please complete the following information):**
- OS: Android

**Smartphone (please complete the following information):**
- Device: Samsung Galaxy A53
- OS: Android
- Version 14

juniorsaraviao, 2024-03-18:
any updates related to this issue?

pwespi, 2024-04-06:
I cannot reproduce this on my test devices. Were you able to reproduce this on multiple different devices?

peitschie, 2024-04-06:
Just passing through... I've have seen this with some Android phones (specifically with a Nokia device on Android 10). No idea why some phones allow it... But it never seemed to me to be a plugin kind of bug.

If the firmware and hardware in Android allow it... Should the plugin really be concerned about preventing this?

juniorsaraviao, 2024-04-16:
Hi everyone, I could replicate the same issue using a Galaxy A11 physical device

ehsieh-semios, 2024-05-21:
I recall this is a defect from the firmware (seen in Samsung phones), maybe you can try to "normalize" it by `off and on before the next scanning`

abdh-dev, 2024-08-27:
Oh yeah this also happened to me with my samsung galaxy a12 phone, at first it was unusual but i never paid much attention to it until i found this issue